openscad/openscad #3301: Strange preview with linear_extrude

kjiwa, 2020-05-06:
Hello. I'm seeing some strange behavior with the following design.

```
difference() {
  linear_extrude(height=10) {
    difference() {
      square(10, center=true);
      square(5, center=true);
    }
  }

  cube(1, center=true);
}
```

The preview ends up having some clipping issues (I'm not sure what the proper terminology is for this).

[image]

The design renders fine though.

[image]

The problem seems to start with that cube at the end. It can be any shape, but everything renders just fine until it is added.

I tested this using the latest development snapshot 2020.04.26.ci4968 compiled for x64. The same behavior exists in the stable build, 2019.05.

MichaelAtOz, 2020-05-06:
Try `linear_extrude(height=10, convexity=10)`

kjiwa, 2020-05-06:
Thanks, that helped.

closed 2020-05-06 by kjiwa