openscad/openscad #3302: Chinese (Traditional) Translation inquiry

microwave54, 2020-05-06:
Hello,
  I'm Michael Tsao, a staff member of the Open Source Software Project under the supervision of the Ministry of Education in Taipei, Taiwan. I've written an email to info@openscad.org on 4/30/20, but haven't received an reply so I'm hoping I'll have better luck here.

My team and I are compiling a list of open-source software to localize for the Taiwanese community and think OpenSCAD is a great software.

From the locale folder on github, it doesn't look like anyone has created a zh_tw.po file. I just want to confirm there is no one currently translating the software interface into Chinese (Traditional) language.

If not, my team would be interested in the Translation role.

Would anyone be able to help us implement the Chinese (Traditional) language translation file into the software once we've completed the translations or how does it work?

Hope to hear back from y'all soon. Have a great day!

Kind regards,
  Michael

-----
Chinese Culture University, School of Continuing Education
曹晉維 (Michael Tsao)
E-mail：cwtsao@sce.pccu.edu.tw
Address：No. 231, Section 2, Jianguo South Road, Da’an District, Taipei City, 10659
Tel：(02)2700-5858 ext. 8630

t-paul, 2020-05-06:
I'm not aware of any ongoing translation activity, so the current status is what you have already seen in https://github.com/openscad/openscad/tree/master/locale + an open PR #3289 for Armenian language. In case anyone is working quietly on this, maybe they come forward based on the discussion here.

As soon as the translation file is available, we can integrate it which will make it usable in the nightly builds, there should be no additional coding required. It may need some additional tests if there's any strange effects in the GUI but those can be reported as normal bugs.

microwave54, 2020-05-13:
Hi @t-paul,
  My team have decided to help translate Openscad into Chinese (Traditional) language. However, we have to a few questions before we can proceed.

1. When we complete the translation, would you be the contact person to contact and do I just post the download link to the .po file here or create a pull request like PR [#3289](https://github.com/openscad/openscad/pull/3289)?

2. When we complete the translation, could it be included in the next release?

3. When will the next release come out?

4. In the new release could you please put the following in the **About** page under **Translation**: 
Distance Learning Center at the Chinese Culture University (Open Source Software for School project) 

Thank you so much for your time and hope to hear back soon!

t-paul, 2020-05-13:
Thanks @microwave54, that's great to hear.

1) Yes, a PR would be best, should there be any issues creating one, we can find a way via download link too.

2) Yes, it would be in the next release

3) Giving dates is difficult as everyone is working on available spare time. The plan was to release a version at the beginning of this year which did not happen. There's still some open topics to wrap up. We hope to get into Release-Candidate mode some time in the next 2 month.

4) Sure, we are happy to add that (or just add via PR too)

microwave54, 2020-05-28:
Hi @t-paul,
  Here is a full Chinese (Traditional) translation ([zh_TW.po file](
https://drive.google.com/file/d/1yYw6bvxElU4USFbin8LemS3GKIeY65-I/view?usp=sharing)
) for OpenSCAD. Sorry, I don't know how to create a PR yet and need your help to add the file into the software.
  Could you please send us the OpenSCAD x86 (64-bit) - exe installer or link to the .exe file so we can check if there is any errors?

My email is E-mail：cwtsao@sce.pccu.edu.tw

Sorry for the hassle. Thank you so much!

t-paul, 2020-05-28:
Testing on Linux using
```
LANG=zh_TW.UTF-8 ./OpenSCAD-2020.05.28.ai5218-_add-translationd84ddfb-x86_64.AppImage
```
looks great.

I just found out a couple of days ago when looking at the other translation PR that the template was not fully up to date with the latest GUI changes. I'm sorry about that. All updated in the PR #3335 now. It would be great if you could have another look at file.
Please review also PR #3335 regarding about window updates and the changes in the `zh_TW.po` header as there were a couple of build warnings. I've added the information based on your initial post here. 

Win32: [Installer](https://5216-1049088-gh.circle-artifacts.com/0/32-bit/OpenSCAD-2020.05.28.ci5216-x86-32_add-translation-Installer.exe) / [ZIP](https://5216-1049088-gh.circle-artifacts.com/0/32-bit/OpenSCAD-2020.05.28.ci5216-x86-32_add-translation.zip)
Win64: [Installer](https://5217-1049088-gh.circle-artifacts.com/0/64-bit/OpenSCAD-2020.05.28.ci5217-x86-64_add-translation-Installer.exe) / [ZIP](https://5217-1049088-gh.circle-artifacts.com/0/64-bit/OpenSCAD-2020.05.28.ci5217-x86-64_add-translation.zip)
Linux: [AppImage 64bit](https://5218-1049088-gh.circle-artifacts.com/0/64-bit/OpenSCAD-2020.05.28.ai5218-_add-translationd84ddfb-x86_64.AppImage)

[image]

closed 2020-06-06 by t-paul

microwave54, 2020-07-15:
Hi @t-paul 
  My appologize for getting back to you sooner. We've been caught up with the midterm evaluation for the Open Source Project for the Ministry of Education. 
  I've reviewed the zh_TW.po and everything looks good. When do you think the latest release will occur? I've remembered you mentioned in the previous message that it might happen sometime in July.
  In the meanwhile, could you please share the link to the following for me again:
Win32: Installer / ZIP
Win64: Installer / ZIP
Linux: AppImage 64bit

Do you mind if we put these version up on our website for now? When the release comes out we will link it back to the Openscad official download page. Thank you so much and have a wonderful evening!

t-paul, 2020-07-15:
This change is now in all the official [snapshots releases](https://www.openscad.org/downloads.html#snapshots), including [Snap nightly](https://snapcraft.io/openscad-nightly) which are used by many people to stay closer to latest developments. The snapshot releases can be installed in parallel to the release version, as in some cases it's useful to reference against the release.

Link to the (automatically updated) snapshot releases mentioned above is: https://www.openscad.org/downloads.html#snapshots

For release things have not progressed much unfortunately as people don't seem to have much time to spare lately. I still hope it's not too far out, but this month is probably not realistic anymore.

microwave54, 2020-07-16:
Thank you so much for the link to the snapshots releases. We'll provide the download links on our website and start promoting Openscad to K-12 teachers and students in Taiwan.

No rush on the official release! It's been a pleasure to corresponding with you and appreciate all the help!
Stay safe and have a great day!

commit 80376ce1 (2020-06-06): Add translation (closes #3302).

Translation provided by Distance Learning Center at the Chinese Culture
University (Open Source Software for School project).