openscad/openscad #3304: $fn option for rotate_extrude?

JasonWandel, 2020-05-07:
I've just started using this tool (it's wonderful!), but can see a discrepancy in the doco. $fn is present in the examples for rotate_extrude but not listed in the parameters list like for cylinder or linear_extrude.

{I know I could update the doco myself, but I'm currently having chemo and can't get my head around what the change needs to be...)

TIA!

MichaelAtOz, 2020-05-07:
Done.

closed 2020-05-07 by MichaelAtOz

nophead, 2020-05-07:
I am not sure it makes sense to list special variables as parameters because you don't need to pass them as a parameter and it appears you can pass any special variable to any module. For example:
```
translate([1,2], $fn = 3)
    circle(10);
```
Draws a triangle.

JasonWandel, 2020-05-07:
Actually, on reflection, I think you're right, They should be listed in a separate section.

Having only just started with openSCAD less than 2 days ago, one thing that's struck me is how similar this language seems to Verilog, and IIRC Verilog allows special parameters to be listed anywhere (I'm a little rusty on it since I've not done anything with Verilog since working on simulators in the 90's).

MichaelAtOz, 2020-05-07:
I just replicated what was shown for `cylinder()`, `sphere()` & `circle()`, I think it is good to show as a hint on how to handle smoothing the `rotate_extrude()`. It applies to all the curvy operations.

nophead, 2020-05-07:
It does need something to show which modules are affected by them but they
aren't parameters with default values as shown for cylinder. They have
global defaults.

Also rotate_extrude will also be affected by $fa and $fa if $fn is 0, as
all curves are. I think even the curves generated by offset.

MichaelAtOz, 2020-05-07:
I included $fa & $fs. Feel free to edit.

JasonWandel, 2020-05-07:
One thing that should perhaps be mentioned if they are broken out into a separate section, is to discuss the implications of high values a bit more. I started a render of a 100-segment circle extruded in 10000 steps (I'd inadvertently pressed the 0 an extra time, meant to be 1000)... 6 hours later my computer locked up entirely!