openscad/openscad #3307: Can't open file directly from finder (macOS)

Fr3d-P, 2020-05-09:
On macintosh, if I double-click a .scad file, the Finder launch openSCAD, but I end-up in the welcome... window when "show welcome screen" is checked in preferences.

The welcome window should appear only when opening openSCAD directly, not when opening a file (more that that, in this window, I don't see any way to continue opening this file, I have to select it again after clicking on  the button "open".

t-paul, 2020-05-09:
Which version is it? If possible please also check with the latest development snapshot.

jarekmodra, 2020-05-15:
I'm getting the same issue with 2019.05 and 2020.05.12

vincentl, 2020-05-15:
If openSCAD is already running (even if the welcome screen is open), double clicking a file in the Finder correctly opens the file in openSCAD.

Guessing that the QEvent::FileOpen event arrives before the SCADEventFilter is registered, but haven't been able to validate that yet.

linuxdan, 2021-03-06:
FWIW, I am using macOS 10.15.7 and OpenSCAD 2021.01 and double-clicking a .scad file works just fine.

3Dietrich, 2022-01-20:
I am using macOS 12.0.1
in OpenScad version 2021.01 
AND OpenSCAD version 2022.01.20

I confirm this bug. @kintel

kintel, 2022-01-20:
Thx, I can reproduce this as well

kintel, 2024-12-03:
Closing as duplicate of #2682

closed 2024-12-03 by kintel