openscad/openscad #3309: Improve preview rendering performance (Implement VBO renderer)

oysteinkrog, 2020-05-11:
I'm creating this issue so I can post a bounty on completing the WIP VBO renderer in https://github.com/openscad/openscad/pull/1935

oysteinkrog, 2020-05-11:
https://www.bountysource.com/issues/91399377-improve-preview-rendering-performance-implement-vbo-renderer

MichaelAtOz, 2020-05-12:
I added $50 from the OpenSCAD pool.

mneuhaus, 2020-10-22:
i tossed in $50 as well 😄

KjellMorgenstern, 2021-11-01:
Since #1935 has been merged to master, I think the OpenScad team can claim the bounty.

t-paul, 2021-11-01:
Thanks for the ping, closing this issue as #1935 is merged. This should make it available for claiming on Bountysource (sometimes takes a few hours to sync).

Most of the work done by @obliviand, please claim the bounty :tada:.

closed 2021-11-01 by t-paul