openscad/openscad #3312: Preview renders intersection cube(0) intersection incorrectly

bldewolf, 2020-05-12:
While experimenting with manipulating empty objects, I tried using `cube(0)` and discovered the preview renders this code incorrectly:
```
intersection() {
cube(0);
cube(1);
cube(5);
cube(10);
}
```

It shows a `cube(5)` instead of an empty object.  The Render version shows the correct object.

Discovered in version 2019.05.

thehans, 2020-05-28:
I can see the source of the issue here, which is partly due to this section of code in CSGTreeEvaluator::applyToChildren:
https://github.com/openscad/openscad/blob/fb349cc1f22e70cf9733260918c4a8f1e844d479/src/CSGTreeEvaluator.cc#L60-L66  
Basically if the left node `t1` in the CSG tree is null then it just replaces it with the right node `t2` and keeps processing through the tree.  This is ok behavior for union operations but not intersection or difference (though it only matters if the *first* term is null for difference).

But I ran into a problem when trying to fix it: basically OpenSCAD doesn't differentiate between "null/empty geometry" and "not even considered geometry", both of them just end up being null `shared_ptr<CSGNode>`.  

For example, my first attempt to fix this failed to render the following test case snippet, since `echo(...);` creates an empty group node, which ends up being a null `shared_ptr<CSGNode>`:
https://github.com/openscad/openscad/blob/fb349cc1f22e70cf9733260918c4a8f1e844d479/testdata/scad/2D/features/intersection2-tests.scad#L12-L17
So that part of the test script is expected to render something but it gets omitted as empty geometry.

t-paul, 2020-10-02:
Fixed via #3342.

closed 2020-10-02 by t-paul