openscad/openscad #3313: openSCAD: weird rendering interaction

pculverhouse, 2020-05-14:
ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation! Expr: e->incident_sface() != SFace_const_handle() File: /Users/kintel/code/OpenSCAD/libraries/install/include/CGAL/Nef_S2/SM_const_decorator.h Line: 330

I have a STL for a camera (Linea_ML_8k below).
When I render it (F6) by itself it is fine , but adding a cube to the mix causes a fault in rendering.

Any ideas?

module Linea_ML_8k(X,Y,Z){ // Dalsa linea-ML 8K colour camera
    translate([X,Y,Z]) rotate([0,0,0]) 
    import ("./03-129-20204-03_LINEA_ML_&_HS_8K_USER_3D_MODEL_low.stl");
}

Linea_ML_8k(0,0,-30);
//cube([40, 40, 10]);

//=========== render listing output

OpenSCAD 2019.05
https://www.openscad.org/

Copyright (C) 2009-2019 The OpenSCAD Developers

This program is free software; you can redistribute it and/or modify it under the terms of the GNU General Public License as published by the Free Software Foundation; either version 2 of the License, or (at your option) any later version.

Saved design '/Users/culverhouse/Documents/SCAD/Dalsa_lineaML-8k-May-2020/MK-8k-issue.scad'.
Loaded design '/Users/culverhouse/Documents/SCAD/Dalsa_lineaML-8k-May-2020/MK-8k-issue.scad'.
Compiling design (CSG Tree generation)...
Compiling design (CSG Products generation)...
Geometries in cache: 27
Geometry cache size in bytes: 13740432
CGAL Polyhedrons in cache: 17
CGAL cache size in bytes: 19490992
Compiling design (CSG Products normalization)...
Normalized CSG tree has 2 elements
Compile and preview finished.
Total rendering time: 0 hours, 0 minutes, 0 seconds

Parsing design (AST generation)...
Compiling design (CSG Tree generation)...
Rendering Polygon Mesh using CGAL...
ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation! Expr: e->incident_sface() != SFace_const_handle() File: /Users/kintel/code/OpenSCAD/libraries/install/include/CGAL/Nef_S2/SM_const_decorator.h Line: 330
Geometries in cache: 30
Geometry cache size in bytes: 19928496
CGAL Polyhedrons in cache: 18
CGAL cache size in bytes: 33025896
Total rendering time: 0 hours, 0 minutes, 9 seconds
Top level object is a 3D object:
Simple: yes
Vertices: 7611
Halfedges: 30874
Edges: 15437
Halffacets: 15844
Facets: 7922
Volumes: 5
Rendering finished.

Saved design '/Users/culverhouse/Documents/SCAD/Dalsa_lineaML-8k-May-2020/MK-8k-issue.scad'.
Loaded design '/Users/culverhouse/Documents/SCAD/Dalsa_lineaML-8k-May-2020/MK-8k-issue.scad'.
Compiling design (CSG Tree generation)...
Compiling design (CSG Products generation)...
Geometries in cache: 30
Geometry cache size in bytes: 19928496
CGAL Polyhedrons in cache: 18
CGAL cache size in bytes: 33025896
Compiling design (CSG Products normalization)...
Normalized CSG tree has 2 elements
Compile and preview finished.
Total rendering time: 0 hours, 0 minutes, 0 seconds

Saved design '/Users/culverhouse/Documents/SCAD/Dalsa_lineaML-8k-May-2020/MK-8k-issue.scad'.
[... 346 lines cut ...]

Saved design '/Users/culverhouse/Documents/SCAD/Dalsa_lineaML-8k-May-2020/MK-8k-issue.scad'.
Loaded design '/Users/culverhouse/Documents/SCAD/Dalsa_lineaML-8k-May-2020/MK-8k-issue.scad'.
Compiling design (CSG Tree generation)...
Compiling design (CSG Products generation)...
Geometries in cache: 33
Geometry cache size in bytes: 25021440
CGAL Polyhedrons in cache: 20
CGAL cache size in bytes: 46572144
Compiling design (CSG Products normalization)...
Normalized CSG tree has 1 elements
Compile and preview finished.
Total rendering time: 0 hours, 0 minutes, 0 seconds

Parsing design (AST generation)...
Compiling design (CSG Tree generation)...
Rendering Polygon Mesh using CGAL...
Geometries in cache: 33
Geometry cache size in bytes: 25021440
CGAL Polyhedrons in cache: 20
CGAL cache size in bytes: 46572144
Total rendering time: 0 hours, 0 minutes, 0 seconds
Top level object is a 3D object:
Facets: 55503
Rendering finished.

pculverhouse, 2020-05-14:
I forgot the STL. see attached.

03-129-20204-03_LINEA_ML_&_HS_8K_USER_3D_MODEL_low.stl.zip
[03-129-20204-03_LINEA_ML_&_HS_8K_USER_3D_MODEL_low.stl.zip](https://github.com/openscad/openscad/files/4628708/03-129-20204-03_LINEA_ML_._HS_8K_USER_3D_MODEL_low.stl.zip)

t-paul, 2020-05-14:
I suspect you closed by accident?

It's normal that STL alone is fine, as this will not be processed in any way. Only the added `cube`  triggers the processing steps which then highlights the issue.

This message means there is likely problems with the STL, or it causes problems in OpenSCAD if it is high resolution with lots of very close vertices.

pculverhouse, 2020-05-14:
Correct! a sad newbie.

Thanks. I'll get back to the suppliers. I've tried Blender and Meshlab to fix the issue.

Phil

t-paul, 2020-05-14:
For MeshLab, there's a pointer int the [FAQ](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#Why_is_my_imported_STL_file_appearing_with_F5_but_not_F6?) how to fix one common class of issues.

pculverhouse, 2020-05-14:
Thanks. I've tried that. I'm just looking at meshmixer now.

t-paul, 2020-05-14:
Hum, that STL crashed my blender (2.82 on Linux), that's not a good sign.

pculverhouse, 2020-05-14:
I can read it in blender. I'll give you the original. I posted the low resolution conversion from STP.
[03-129-20204-03_LINEA_ML_&_HS_8K_USER_3D_MODEL.stl.zip](https://github.com/openscad/openscad/files/4628856/03-129-20204-03_LINEA_ML_._HS_8K_USER_3D_MODEL.stl.zip)

pculverhouse, 2020-05-14:
Well, meshmixer sorted it. I ran through the cleaning options then exported as solid. It takes forever to render, bu it works.

Thanks for you ideas and support.

Phil

closed 2020-05-14 by pculverhouse