openscad/openscad #3314: Build issue against boost-1.73.0 on Gentoo Linux

waebbl, 2020-05-15:
Boost library has removed the `boost/detail/endian.hpp` include file with version 1.73.0. Therefore the build of `src/import.cc` fails with this version of boost and needs to be replaced with `boost/endian/arithmetic.hpp`.

The patch https://github.com/gentoo/gentoo/pull/15809/files#diff-1a5e311f8dc572c6a9925127202763bb solves the issue on Gentoo.

t-paul, 2020-05-15:
I've looked into that. I think we may want to discuss going to boost-1.58 which I think is the first one that would not need deprecated stuff.

t-paul, 2020-05-26:
Looks like the `endian` defines are [deprecated already](https://groups.google.com/forum/#!topic/boost-developers-archive/vCOc8F97wXc), so it looks like moving to the `predef` defines is the way to go. That makes us dependent on boost-1.55.

hroncok, 2020-06-01:
Trying with https://github.com/openscad/openscad/commit/6fecf31de12e74f4c63b146365a69ce567ef64dc

hroncok, 2020-06-01:
Still fails with: boost/detail/endian.hpp: No such file or directory

t-paul, 2020-06-01:
How? Where is that coming from?
``` 
openscad-branch $ git checkout 6fecf31
HEAD is now at 6fecf31de Update to boost-1.55 (fixes #3314).
openscad-branch $ grep -ir endian.hpp
openscad-branch $
```

hroncok, 2020-06-01:
Sorry for not being verbose enough. I've applied 6fecf31 on the latest stable release.

t-paul, 2020-06-01:
Right, that line was removed earlier as it's not actually used. So unfortunately there's no dedicated commit that could be just moved back to the release version.

I think it might be enough to really just remove that line to make it compatible with new boost. This is in file `import.cc` where the actual endian usage is in `import_stl.cc` where it's likely coming from the Predef lib for a while now.

We don't have any big endian test runs right now, that would be nice to have. Does Fedora builds run the test suite at some point on big endian machines?

hroncok, 2020-06-01:
> Does Fedora builds run the test suite at some point on big endian machines?

Yes, on s390x. But for the lack of my time, the results are informative only (failures don't abort the build). 

See https://koji.fedoraproject.org/koji/search?terms=openscad&type=package&match=exact

click a build (such as openscad-2019.05-10.fc33) and click build.log in the Logs/s390x section.

https://kojipkgs.fedoraproject.org//packages/openscad/2019.05/10.fc33/data/logs/s390x/build.log

28% tests passed, 892 tests failed out of 1244 :(

t-paul, 2020-06-01:
Thanks, I'll have a look at the logs. Considering this change specifically should only affect binary STL import, I would assume seeing a bug report relatively quickly if it's broken and actually used. Looking at the new code coverage report it seems we do have at least some binary STL test in the test suite.

Maybe we can find a way to run some big endian build in the CI pipeline at some point.

t-paul, 2020-06-01:
That s390x run seems to fail pretty much everything where image output is generated, so it might be some OpenGL issue. It would be interesting to see the generated error report.

hroncok, 2020-06-03:
We've added:

 - [boost-1.73.patch](https://src.fedoraproject.org/rpms/openscad/blob/2a247b23cf1370662b5e537b196d0acb09bdbb7e/f/boost-1.73.patch)
 - [openscad-missing-include.patch](https://src.fedoraproject.org/rpms/openscad/blob/2a247b23cf1370662b5e537b196d0acb09bdbb7e/f/openscad-missing-include.patch)

hroncok, 2020-06-03:
> It would be interesting to see the generated error report.

I can try to package it in the resulting RPM.

commit 6fecf31d (2020-06-06): Update to boost-1.55 (fixes #3314).

- Move to Predef macros which are available since 1.55
- Boost endian macros are deprecated as of 2018-09-03, see
  https://groups.google.com/forum/#!topic/boost-developers-archive/vCOc8F97wXc
closed 2020-06-06 by t-paul