openscad/openscad #3315: Do not constrain the precision of manually entered numeric values in customizer

tturpin, 2020-05-17:
Somehow related to #3098: When a variable has no specified range and precision, it is impossible to specify a value with more significant digits than the default value, which is an annoying limitation.
I understand than the increase/decrease actions need a step value, but the text field should allow an arbitrary precision.

Possibly, the number of digits in the default value literal could be used instead of the significant (i.e., non-zero) digits to guess the precision, so that we may use for example a default value of 1.00 (and not 1.01) to allow two digits after the point.

buteomont, 2020-07-16:
I too would like to see this problem fixed.

JeffreyRodriguez, 2020-12-24:
Present workaround is to set defaults like "1.0001"

It works, but it's a lot uglier than just being able to punch in an arbitrarily precise value.