openscad/openscad #3317: Crash on start: dbus_connection_unref() bug

BlueNalgene, 2020-05-17:
Version: OpenSCAD version 2020.05.16 (nightly snap)
OS: Ubuntu 18.04 Bionic (running KDE 5.44.0)

Problem: The program crashes on start.  Running in terminal throws an error message.

Error message:

> The last reference on a connection was dropped without closing the connection. This is a bug in an application. See dbus_connection_unref() documentation for details.
Most likely, the application was supposed to call dbus_connection_close(), since this is a private connection.
  D-Bus not built with -rdynamic so unable to print a backtrace
Aborted

BlueNalgene, 2020-05-17:
OpenSCAD runs properly when compiled from source.  There may be a problem with the Snap package since the error message is similar to [https://github.com/snapcore/core18/issues/4](https://github.com/snapcore/core18/issues/4).

t-paul, 2020-05-17:
Yep, I found that too, trying to revert back to old core16.

t-paul, 2020-05-18:
Pushed an update to https://snapcraft.io/openscad (which is also affected) so if the build succeeds that should show up in the edge channel in an hour or so. If that works, I'll update the nightly too.

closed 2020-05-18 by t-paul

t-paul, 2020-05-18:
ARM64 is building for hours now, so I published everything else for https://snapcraft.io/openscad-nightly - I guess we have to find out what's going on with `core18`. It seems strange that this does not work on 18.04 but works fine on Debian/Testing.

t-paul, 2020-05-18:
Ah, ARM64 finally finished too, so everything release. On the plus side, both versions now have 3MF support enabled :grinning:.

BlueNalgene, 2020-05-18:
Confirmed.  Latest Snap is working.  Thanks.

commit 92a924a8 (2020-05-18): Revert to Ubuntu 16.04 based core (fixes #3317).