openscad/openscad #3318: find search word small glitch: character offset(related to international characters in comments)

kipbits, 2020-05-18:
I've noticed highlight match words seams to be having a glitch in the latest public build(2019.05). it appears to be finding the right lines, although the actual text highlighted seams to be offset by some consistent number of characters going down the file(aside from the first occurrence of the word). ...oddly after testing other files, the problem is hard to reproduce, although copy/pasting my text into a new file duplicates the problem. ...on further investigation it seams the amount of offset is related to the number of times my file contains the character 'ø' within my comments. It's been awhile since I've worked on the QScintilla editor, although I guess i didn't think to type in any funny characters back then. Why should this be happening? I don't think it is a feature; perhaps I like to type greek characters in my comments: α, β, γ, δ, ϵ...π... it appears any character above the basic ascii set gives the same result...

[image]

Anyway, seams like something that should be fixed... ...and maybe I will look into it, although at the moment I'm sort of tied up dealing with all sorts of crap in my life, not to mention my finger is still buggered up. And apparently i have no legal recourse for any of this; maybe I just haven't filled out the right form yet... Anyway I'll have to dig up my computer i used last time, up out of the rubble and update my build. ...it may take me some time ;)

t-paul, 2020-05-18:
Looks like a duplicate of #2795. Could you check with a [snapshot build](https://www.openscad.org/downloads.html#snapshots)?

kipbits, 2020-05-18:
ok, never mind, looks like it's already fixed. ...sorry for duplicate.

closed 2020-05-18 by kipbits