openscad/openscad #3319: BOOST_SP_TYPEID： internal compiler error

PuddingPengChen, 2020-05-20:
Hi，I am trying to use MXE to compile openscad from source code in Ubuntu18 , but when I execut make, there is an error:
`  from src/csgnode.cc:30:
/home/binro/mxe/usr/i686-w64-mingw32.static/include/boost/smart_ptr/shared_ptr.hpp: In function 'D* boost::detail::basic_get_deleter(const boost::shared_ptr<U>&)':
/home/binro/mxe/usr/i686-w64-mingw32.static/include/boost/smart_ptr/shared_ptr.hpp:932:54: internal compiler error: in typeid_ok_p, at cp/rtti.c:319
     return static_cast<D *>( p._internal_get_deleter(BOOST_SP_TYPEID(D)) );
                                                      ^
Please submit a full bug report,
with preprocessed source if appropriate.
`
 Has any one ever run into this problem?

t-paul, 2020-05-20:
No, I have not seen that. Which compiler version is that? We are using `gcc7`.

Compile options we use for the official builds can be seen in:

- gcc7 base image: [Dockerfile](https://github.com/openscad/docker-openscad/blob/master/mxe/mxe-x86_64-deps/Dockerfile)
- remaining dependencies: [Dockerfile](https://github.com/openscad/docker-openscad/blob/master/mxe/mxe-x86_64-deps/Dockerfile)

donbright, 2020-05-21:
this would also possibly depend on the boost version MXE is using since it's dying inside of boost

PuddingPengChen, 2020-05-23:
hi,t-paual,
my gcc version：gcc (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0. And also, I haved tried gcc5.5,  they all have the same problem.  I input "i686-w64-mingw32.static-gcc --version" and get the mingw-gcc version is:5.5.0.  Is the gcc7 means "mingw32-gcc7"?

t-paul, 2020-05-23:
To get the MINGW GCC7, you'll have to add `MXE_PLUGIN_DIRS=plugins/gcc7` to all the `make` calls as can be seen in the linked `Dockerfile`s above. If that does not help, I'm out of ideas. It's probably worth reporting at the MXE repo.

PuddingPengChen, 2020-05-25:
hi,t-paual,
I solved it after add `MXE_PLUGIN_DIRS=plugins/gcc7`. Thank you very much!  Please close the issue :+1:

closed 2020-05-25 by t-paul