openscad/openscad #3320: Occasional SIGSEGV (signal 11)

d-a-v, 2020-05-20:
Hi,

I am an occasional user, using openscad-nightly and I run into a segfault under linux while rendering.
I recompiled git-master to get more details on the error, which are reproduced below.
Source code producing the issue is provided.
The included variable `fn` is set to 50. Strangely when set to 100, there is no error.
I found that odd enough to post this issue.

I was not able to decide if the message `PolySet has nonplanar faces. Attempting alternate construction` was telling me that unpredictable behavior could possibly be my fault.

```
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
==7183== Invalid read of size 8
==7183==    at 0x130457A: CGAL::HalfedgeDS_halfedge_base_base<CGAL::HalfedgeDS_list_types<CGAL::Cartesian<CGAL::Gmpq>, CGAL::I_Polyhedron_derived_items_3<CGAL::Polyhedron_items_3>, std::allocator<int> >, CGAL::Boolean_tag<true>, CGAL::Boolean_tag<true>, CGAL::Boolean_tag<true> >::opposite() (HalfedgeDS_halfedge_base.h:55)
[...]
==7183==  Address 0x0 is not stack'd, malloc'd or (recently) free'd
==7183== 
==7183== 
==7183== Process terminating with default action of signal 11 (SIGSEGV)
```

[segfault.scad.txt](https://github.com/openscad/openscad/files/4656421/segfault.scad.txt)

MichaelAtOz, 2020-05-20:
Render on Windows7 crashes, 2019.05 & 2020.05.06.

MichaelAtOz, 2020-05-20:
Simplified code still crashes.
```
difference()
{
      
        polycyl(r=r, h=h, vstep=2, p=0, fn=fn);
        translate([0,0,1/16])
        polycyl(r=r-thick, h=h, vstep=2, p=0, in=1, fn=fn);
}
```
[image]
This is the negative of the difference.
Ignore the small cube, it's for testing. 
Note that peak is at z=0.
OpenSCAD shouldn't crash, but putting an edge on a face is not a valid.
But the `translate` above moves it away, it still crashes...
I exported the two polycyl's (separately).
The negative is bad. It has holes & self intersections.
[image]

d-a-v, 2020-05-21:
Thanks for your analyse !

I must say I was not able to get what you are showing in the first image.
Can you specify the name of the tool you are showing below? Maybe I would be able to come and bother you with real problems if I could use it.

I changed the module so to work only with integer indices and with triangles as faces. 

Here is a much simple intersection with two 18-faces-as-triangles-objects. One is thiner and taller than the other. Ttrying to subtract one from the other, I get no segfault but assertions like this one:
```
ERROR: CGAL error in CGAL_Nef_polyhedron3():
CGAL ERROR: assertion violation!
Expr: e->incident_sface() != SFace_const_handle()
File: /usr/include/CGAL/Nef_S2/SM_const_decorator.h Line: 329 
```
[Source is attached](https://github.com/openscad/openscad/files/4663209/assert2.scad.txt)

t-paul, 2020-05-21:
The crash is a duplicate of #2847 which is difficult to fix as the crash happens inside CGAL. Please update title to cover the other issue or close.

t-paul, 2020-05-21:
Just looking at the last example and using one of the `polycyl()` calls, it seems you have twice as many triangles than needed (24 instead of 12). Some may even have wrong winding order? 
Also there's 2 vertices too much 10 instead of 8 with vertices 0/3 and 4/7 overlapping.

Maybe checkout https://github.com/revarbat/BOSL2/wiki/debug.scad

d-a-v, 2020-05-21:
Closing as duplicate of #2847,
Thanks for the help @MichaelAtOz @t-paul !

closed 2020-05-21 by d-a-v

MichaelAtOz, 2020-05-21:
> I must say I was not able to get what you are showing in the first image.

```
difference()
{
        //polycyl(r=r, h=h, vstep=2, p=0, fn=fn);
        translate([0,0,1/16])
        polycyl(r=r-thick, h=h, vstep=2, p=0, in=1, fn=fn);
}
```
Then look at the base

> Can you specify the name of the tool you are showing below? Maybe I would be able to come and bother you with real problems if I could use it.

That is a Windows tool, [Netfabb](https://www.autodesk.com/products/netfabb/overview?plc=NETFA&term=1-YEAR&support=ADVANCED&quantity=1), mine is Basic which was a 'free' version, I think the current trial version defaults to something similar after the trial.

> I changed the module so to work only with integer indices and with triangles as faces.
>
> Here is a much simple intersection with two 18-faces-as-triangles-objects. One is thiner and taller than the [...]

The bug that segfaulted was another of those CGAL assertions, just not handled properly.
Whenever you have a CGAL assertion it will be either a bad imported STL, or a bad polygon/polyhedron. The assertion is saying your geometry doesn't make sense, above for example the e-> face should match the SFace.

You should always debug polyhedrons. 

e.g.
```
...
            // top closure
            for (a = [0:$fn])
                [ (steps*above) + a + 1, (steps*above) + a, topidx ],
        ];
    polydebug(points);
    color("skyblue",0.5)
    polyhedron(points, faces);
}

module polydebug(p) {
  for( i=[0:len(p)-1]) {
    echo(i,p[i]);
    translate(p[i]) {
      color("black") sphere(0.5, $fn=12);
      translate([-2,2,0]) linear_extrude(height=0.5) text(str(i), size=1.5);
    }
  }
}  
```
Using that on your larger poly shows:
[image]
You have internal faces and points 4 & 7 is probably wrong.

The smaller one is also bad:
[image]
Here is Netfabb of the other:
[image]

d-a-v, 2020-05-22:
I could debug my module thanks to your tips. It is now rendering without error.

> You should always debug polyhedrons. 

Agreed. Maybe OpenSCAD should more aggressively stop when incoherence is detected.
A message to redirect occasional users like me towards such valuable information would be very useful. The debug library proved to be very efficient to help debugging.