openscad/openscad #3321: wrong number of faces in z axis

SebiTimeWaster, 2020-05-21:
a sphere with $fn=16 renders correctly with 16 faces on the xy plane, but it has 9 instead of 8 faces on the z axis:
[image]
OpenJSCAD does not have this bug:
[image]

thehans, 2020-05-23:
Your screenshots appear to be rendering a simple rounded cube, but I cannot be sure since youi dont showl the full object, and you haven't given any example code.  

If your goal really is a rounded cube, I would recommend to place the 8 spheres in their respective corners, and just use `hull` on them, then you don't need to worry about lining up cylinders etc.

Sphere always rounds up the number of vertices for latitude to an even number.
Personally I would also prefer for spheres to come to a point at top and bottom.  But this is the way it has been implemented, and changing fundamental primitives now has potential to cause issues in all previously written scad files which may depend on existing behavior.  So its not likely to change, for the sake of compatibility.  Marking as duplicate of https://github.com/openscad/openscad/issues/462

You may also be interested in checking out https://github.com/openscad/openscad/issues/1983
for some alternative sphere examples which are implemented in userspace.

closed 2020-05-23 by thehans

thehans, 2020-05-23:
Also, for an example of a rounded cube module that uses hull, see example006.scad (releases of OpenSCAD before Feb 2019 won't have this updated code):
https://github.com/openscad/openscad/blob/979ecca204eaf3b0e557824f6887243740396595/examples/Old/example006.scad#L1-L15

All OpenSCAD examples are public domain so feel free to copy/paste into any scripts you want.

SebiTimeWaster, 2020-05-23:
@thehans i built what you see in the picture to illustrate the bug for you. but since you are not willing to fix bugs "for backwards compatibility" it does not matter anyway. 
i will permanently switch to JSCAD, it has much better features anyway.

nophead, 2020-05-23:
I defined my own sphere that replaces the built in one OpenSCAD, you can find it here: https://github.com/nophead/NopSCADlib#Sphere I just rotate extrude a semi circle.

It doesn't need a change to OpenSCAD to have any sphere you like.

t-paul, 2020-05-23:
He closed as duplicate issue, pointing to the older still open one which has excaclty the discussion about risk of breaking existing designs.

There is no point in threatening to use a different tool. If that works better for you, that's fine, we are not trying to sell anything here. Just consider how you would feel if we would constantly change geometry behaviour for existing scripts. This is a difficult decision, sometimes taking long time as people work in their spare time and it's difficult to agree. That said, it may indeed end up not being "fixed" as there are simple ways to generate spheres with other tesselation using a line or 2 of scad code.

SebiTimeWaster, 2020-05-23:
@t-paul I am threatening anyone, i am sure the OpenSCAD project will survive just fine without me, i am just doing it.

And no, it is not a difficult decision. If you have a bug you fix it, release a new major (breaking) version and tell the people about that break in the release notes. That's how it is in every piece of software out there (especially software that is related to programming). That you even discuss about this is a bad sign for the project.

t-paul, 2020-05-23:
No, that's not how it's done for programming languages in general, and there's enough feedback from OpenSCAD users asking to do the same.
If you think discussion is a bad sign, we'll probably never agree on anything.

SebiTimeWaster, 2020-05-23:
you know that i meant discussion about not fixing bugs. but hey, whatever.