openscad/openscad #3324: OFF Export Texture (UV) Coordinates

stuartmscott, 2020-05-22:
Hey all,

I've been using OpenSCAD for a while now to generate 3D models which I export as .off files to be rendered in OpenGL, and this has worked really well!

I would now like to texture these meshes, and while the .off file format supports texture coordinates, OpenSCAD does not export this information.

Relevant File: https://github.com/openscad/openscad/blob/master/src/export_off.cc#L85

Would this be something OpenSCAD would consider supporting in future?

Thanks for your time

t-paul, 2020-05-22:
Can you explain a little bit more in detail how that would be described in the model. So far I've seen UV unwrapping in Blender which is a very manual process, so I'm not sure that would even be usable for simple models without additional input.

stuartmscott, 2020-05-22:
Hey t-paul,

Like you, I have used Blender's UV unwrapping to get this information before, but I switched to OpenSCAD because I was able to express ideas more easily/concisely in code, than point+click+drag.

> how that would be described in the model

By model do you mean the *.scad file, or OpenSCAD's internal representation of the object?

I had envisioned this being only calculated when the object is exported, but you're right that there may be situations where the algorithm may not produce optimal results and therefore the manual processing you mentioned would be needed. But at this point I would be happy if OpenSCAD gave it a good attempt and I cleaned up the .off myself - tedious, but preferable to switching back to Blender.

kintel, 2024-03-31:
Closing as not feasible

closed 2024-03-31 by kintel