openscad/openscad #3327: Feature request: Color axis similar to the axis indicator

larsrc, 2020-05-23:
The little X/Y/Z axis indicator has RGB coloring. Why not make the main axis similarly (though probably with a darker green so it's still visible)? That way you always have an indication of which axis you're looking at.

t-paul, 2020-05-24:
If we do that, it needs to go into the color scheme definition.

robertlugg, 2020-07-29:
@t-paul, the small axis appears to use hard-coded Red, Green, and Blue:
```
  glColor3d(1.0, 0.0, 0.0);
  glVertex3d(0, 0, 0); glVertex3d(10*dpi, 0, 0);
  glColor3d(0.0, 1.0, 0.0);
  glVertex3d(0, 0, 0); glVertex3d(0, 10*dpi, 0);
  glColor3d(0.0, 0.0, 1.0);
  glVertex3d(0, 0, 0); glVertex3d(0, 0, 10*dpi);
```
How should the main axis look?

t-paul, 2020-07-29:
Interesting idea with changing only the center bit. I'm not sure what's better.

We could try moving the 3 colors to the color scheme and draw the whole axis in that color, see how that look. The current axis color would then be used only for the tick marks and numbers.

robertlugg, 2020-07-29:
@t-paul, I'll try that.  It will give me a good reason to learn about themes!  Currently, the color of the mini-axes is NOT configurable.  I assume I need to add additional theme options for the three colors.

t-paul, 2020-07-29:
Yes, that would 3 additional entries in the json files (e.g. in `color-schemes/render/tomorrow.json`)

https://github.com/openscad/openscad/blob/0c2af2defa42c88da1f99e694a5696ecf7e1ffda/color-schemes/render/tomorrow.json#L6-L10