openscad/openscad #3336: Reduce number of faces

espr14, 2020-05-28:
There is only one operator that reduces number of faces which is hull(). But it works on convex objects only. It would be nice to have something like:

`module reduce(min_face_size mfs, max_deviation md);
`
This would remove (merge with neighbor) all faces smaller than `mfs` but only in case they don't cause deviation of surface more than `md` at any place.

This would be handy after use of more operators (union, difference etc.) which can lead to too many small faces that are not necessary. It's also great to do it before export to reduce file size.
For example I do export, load it in Meshmixer, reduce faces to half and the model looks the same.

thehans, 2020-06-04:
Personally I don't see much purpose for such a feature.  If you need simpler model, it would make more sense to adjust $fn, or $fs/$fa to not create so many faces in the first place.

I mean, maybe its useful to simplify an imported STL that wasn't created in OpenSCAD, but I feel like its too rare of a case to bother implementing when other free tools can also do the job.
Do you have any specific example case besides importing, where adjusting $fn etc is not a feasible solution?

espr14, 2020-06-05:
STL are not a problem - I use external tools.
I use $fn, $fs, $fa but they do not apply to union, difference, intersection and scale. These operators will create small faces no matter what. One reason is floating point operations that are not absolutely precise. Unaligned objects can multiply number of faces.
Small faces propagate through the hierarchy generating more and more faces. Since these operators have high time complexity (like O(n^3) or more) every face counts a lot.

nophead, 2020-06-05:
Can you give an example? I only get faces in my design where I specify them. Scale should not create any new ones. Union, difference and intersection only create the expected ones for me.

espr14, 2020-06-05:
```
$fn=32;
scale([10,1,1]) hull() sphere();
```
Scale does not create new ones but due to scale much higher $fn is needed. Scale preserves even the smallest faces so it is a mix of big and small ones. Here the reduce() would be helpful.

espr14, 2020-06-05:
```
render()
for(b=[0,1])
translate([b*.99,b*0.01,b*0.01])
cube();
```
This should be simple cube but due to (simulated) float arithmetic it will generate much more triangles.

nophead, 2020-06-05:
I don't understand why you think the result should be a cube. It is exactly
as you coded it. Two overlapping cubes with the second one slightly offset.
Nothing to do with floating point errors.

dpellegr, 2024-11-23:
There would be a pretty strong use case for this feature now that, thanks to the power of the manifold engine, we can easily perform boolean operations with thousands of objects.
Think about a geometric transformation that moves a "tool object" around an object being cut, similar to a machine tool. You need to take very small steps when moving the tool otherwise you cannot capture the cutting action. However this results in many tiny faces which could be smoothed without loss of accuracy iteratively collapsing edges shorter than a given threshold.
Not sure though if this fits within the OpenSCAD ecosystem.