openscad/openscad #3337: Alternative rotate_extrude

espr14, 2020-05-28:
`rotate_extrude` uses 2D shape in every angle. But this is not great because it creates pair of triangles in the same plane.

[image]

Proposed alternative could create something like this from a circle. This alternates angle for odd and even points so each triangle has it own plane. This makes object more precise with the same amount of triangles.

t-paul, 2020-05-28:
This seems a bit too unspecific to warrant a whole new primitive mostly repeating an existing one. There's a number of examples how to specifically generate differently triangulated spheres via scripts/libraries in user space.

nophead, 2020-05-28:
Also it wouldn't play nice with cylinders and Mnikowski rounding.

espr14, 2020-05-28:
What user space do you mean?
I can create the module that will do this but I need set of points and paths out of a 2D model. Is there a way how to get it?

espr14, 2020-05-28:
In fact, this approach creates nicer surfaces and prevents issues where points of two solid objects touch themselves.
[image]

nophead, 2020-05-28:
If you use them to make rounded corners of boxes, etc. the walls will be
crooked  Cylinder and spheres made with the current rotate_extrude meet up
properly and make exact dimensions.

espr14, 2020-05-28:
`$fn=7`

[image]

nophead, 2020-05-28:
Yes, as I said crooked.

This is created with rotate_extrude and $fn = 16;

[image]