openscad/openscad #3343: OpenScad closes unexpectedly on MacOS Catalina when copying / paste a name in save file

Axicontech, 2020-05-31:
Log:

Process:               OpenSCAD [583]
Path:                  /Applications/OpenSCAD.app/Contents/MacOS/OpenSCAD
Identifier:            org.openscad.OpenSCAD
Version:               2019.05 (2019.05.10)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           OpenSCAD [583]
User ID:               501

Date/Time:             2020-05-31 16:43:57.486 -0300
OS Version:            Mac OS X 10.15.5 (19F96)
Report Version:        12
Anonymous UUID:        C82FBD0E-846E-70E0-E035-CA4FB21E14B9

Time Awake Since Boot: 670 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_PROTECTION_FAILURE at 0x00007ffee8d79f58
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [583]

VM Regions Near 0x7ffee8d79f58:
    MALLOC_SMALL           00007f9cd6800000-00007f9cd7000000 [ 8192K] rw-/rwx SM=PRV  
--> STACK GUARD            00007ffee557a000-00007ffee8d7a000 [ 56.0M] ---/rwx SM=NUL  stack guard for thread 0
    Stack                  00007ffee8d7a000-00007ffee957a000 [ 8192K] rw-/rwx SM=SHM  thread 0

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libobjc.A.dylib               	0x00007fff6760a94f lookUpImpOrForward + 58
1   libobjc.A.dylib               	0x00007fff6760f663 object_getMethodImplementation + 52
2   com.apple.CoreFoundation      	0x00007fff2e8ee3d1 _NSIsNSString + 48
3   com.apple.Foundation          	0x00007fff30fb048d -[NSXPCEncoder _encodeObject:] + 60
4   com.apple.Foundation          	0x00007fff30fb5ded _NSXPCSerializationAddInvocationWithOnlyObjectArgumentsArray + 109
5   com.apple.Foundation          	0x00007fff30fb5cbe -[NSXPCEncoder _encodeInvocationObjectArgumentsOnly:count:typeString:selector:isReply:into:] + 183
6   com.apple.Foundation          	0x00007fff30faeae9 -[NSXPCConnection _sendInvocation:orArguments:count:methodSignature:selector:withProxy:] + 2065
7   com.apple.Foundation          	0x00007fff30fb5bde -[NSXPCConnection _sendSelector:withProxy:arg1:arg2:] + 125
8   com.apple.Foundation          	0x00007fff30fb5b5d _NSXPCDistantObjectSimpleMessageSend2 + 33
9   com.apple.ViewBridge          	0x00007fff614c1af6 __37-[NSRemoteView performKeyEquivalent:]_block_invoke_2 + 662
10  com.apple.ViewBridge          	0x00007fff61457a42 synchronizedAuxiliaryProxyFor + 134
11  com.apple.ViewBridge          	0x00007fff614579ad auxiliaryProxyFor + 446
12  com.apple.ViewBridge          	0x00007fff614b6707 -[NSRemoteView _auxiliaryProxyWithSemaphore:attemptingTo:withCompletion:] + 74
13  com.apple.ViewBridge          	0x00007fff614c1845 __37-[NSRemoteView performKeyEquivalent:]_block_invoke + 116
14  libsystem_trace.dylib         	0x00007fff689deba8 os_activity_apply_f + 66
15  com.apple.ViewBridge          	0x00007fff614b2a04 -[NSRemoteViewSemaphore applyActivity:] + 177
16  com.apple.ViewBridge          	0x00007fff614c176a -[NSRemoteView performKeyEquivalent:] + 228
17  com.apple.AppKit              	0x00007fff2bf21771 -[NSView performKeyEquivalent:] + 189
18  com.apple.AppKit              	0x00007fff2c46bf3c -[NSWindow _commonPerformKeyEquivalent:conditionally:] + 73
19  com.apple.AppKit              	0x00007fff2bf47cbd -[NSWindow keyDown:] + 30
20  com.apple.AppKit              	0x00007fff2be054f9 forwardMethod + 204
21  com.apple.ViewBridge          	0x00007fff614c1cc1 -[NSRemoteView keyDown:] + 163
22  com.apple.AppKit              	0x00007fff2bd7754d -[NSWindow(NSEventRouting) _reallySendEvent:isDelayedEvent:] + 6512
23  com.apple.AppKit              	0x00007fff2bd759b9 -[NSWindow(NSEventRouting) sendEvent:] + 349
24  libqcocoa.dylib               	0x000000010ad03dc1 0x10acd8000 + 179649
25  com.apple.AppKit              	0x00007fff2bf226b7 -[NSMenu _performKeyEquivalentWithDelegate:] + 134
26  com.apple.AppKit              	0x00007fff2bf22880 -[NSMenu _performKeyEquivalentWithDelegate:] + 591
27  com.apple.AppKit              	0x00007fff2bf22222 -[NSMenu performKeyEquivalent:] + 71
28  com.apple.AppKit              	0x00007fff2c35817b routeKeyEquivalent + 521
29  com.apple.AppKit              	0x00007fff2bf47cf2 -[NSWindow keyDown:] + 83
30  com.apple.AppKit              	0x00007fff2be054f9 forwardMethod + 204
31  com.apple.ViewBridge          	0x00007fff614c1cc1 -[NSRemoteView keyDown:] + 163
32  com.apple.AppKit              	0x00007fff2bd7754d -[NSWindow(NSEventRouting) _reallySendEvent:isDelayedEvent:] + 6512
33  com.apple.AppKit              	0x00007fff2bd759b9 -[NSWindow(NSEventRouting) sendEvent:] + 349
[... 921 lines cut ...]
__OBJC_RW                         1892K        2 
__TEXT                           309.1M      364 
__UNICODE                          564K        1 
mapped file                      744.8M      280 
shared memory                      632K       14 
===========                     =======  ======= 
TOTAL                              2.2G     1295 
TOTAL, minus reserved VM space     1.9G     1295 

Model: iMac18,3, BootROM 183.0.0.0.0, 4 processors, Quad-Core Intel Core i5, 3,8 GHz, 8 GB, SMC 2.41f2
Graphics: kHW_AMDRadeonPro580Item, Radeon Pro 580, spdisplays_pcie_device, 8 GB
Memory Module: BANK 0/DIMM0, 4 GB, DDR4 SO-DIMM, 2400 MHz, 0x802C, 0x344154463531323634485A2D3247334532202020
Memory Module: BANK 1/DIMM0, 4 GB, DDR4 SO-DIMM, 2400 MHz, 0x802C, 0x344154463531323634485A2D3247334532202020
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x16F), Broadcom BCM43xx 1.0 (7.77.111.1 AirPortDriverBrcmNIC-1610.1)
Bluetooth: Version 7.0.5f6, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en1
Serial ATA Device: APPLE HDD ST2000DM001, 2 TB
USB Device: USB 3.0 Bus
USB Device: USB3.0 Hub
USB Device: My Book 25EE
USB Device: Bluetooth USB Host Controller
USB Device: FaceTime HD Camera (Built-in)
USB Device: Back-UPS CS 650 FW:817.v9.I USB FW:v9
USB Device: USB2.0 Hub
Thunderbolt Bus: iMac, Apple Inc., 41.4

mh-, 2020-11-16:
This also happens e.g. in the Export STL dialog.
OpenSCAD version 2019.05 
macOS Catalina 10.15.7 (19H15)

luxlux, 2021-06-02:
it also crashes ever time when (copy and) PASTE something into the a file name field in the "save as" dialog on macOS Big Sur (11.2.3) / M1 Processor.

t-paul, 2021-11-05:
Please check with the latest snapshot build, this requires macOS 10.13 (High Sierra) and updates some components hopefully fixing the issue.

ChrisCoxArt, 2022-01-16:
I cannot reproduce this in 2021.01 on MacOS 12.1, or current dev code on MacOS 12.1.
The crash log looks like a bug in Apple OS code, probably something failing in a recursive fashion (and eating up the stack).

kintel, 2024-11-28:
No reports like this for a while. Assuming it was fixed.

closed 2024-11-28 by kintel