openscad/openscad #3345: Suggestion / question: Bitwise operator support

samyk, 2020-06-01:
Hi all!

I started working with OpenSCAD but some things have seemed challenging for me to do programmatically and wanted to build some features if it would be considered for master branch. If I built bitwise operators into the language, would that be accepted to master, or should users be using Processing or similar tool for those sorts of things, or am I using OpenSCAD incorrectly or haven't learned the right way to use it?

Bitwise operators were one of the things I immediately ran into that complicated a project.
To implement bitwise AND, I needed to build a recursive and somewhat excessive arithmetic function, and wasteful on large operations since the CPU/GPUs can perform bitwise operations quickly, and I'm specifically using OpenSCAD to perform many-polygon type projects rather than traditional CAD.

```
// 16-bit OpenSCAD AND function -samy kamkar
function and(a, b, c=0, s=0) = 
  s == 15 ? 
  c : 
  and(a, b, 
    (c + ((floor(a/exp(ln(2)*s) % 2) * 
    floor(b/exp(ln(2)*s)) % 2)) *
    exp(ln(2)*s)
  ),
  s+1);
```

The non-return modules and single-expression functions also makes things a bit challenging. Are these restrictions due to some underlying optimization that prevents a more natural, multi-expression subroutine with a return value?

Thanks,
samy

MichaelAtOz, 2020-06-01:
This does not answer your primary question, but have you seen 
https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/List_Comprehensions

samyk, 2020-06-01:
Thank for sharing, I checked the page out but is there something there that would simplify something here? I didn't notice anything.

Another challenge was doing bitwise adjustment on an iterator in a for loop, but as conditionals aren't part of `for` loops and while loops don't exist, again a recursive function is necessary to do a conditional with a non-static change of the iterator.

I'm sure this is for a reason but would love to understand why and if additions of features like this would be accepted.

Thanks!

thehans, 2020-06-01:
Bitwise operations are typically performed on integer types, but OpenSCAD language only has a single numeric type, which is double precision floating point.   So I'm not even sure how you would define such bitwise operations over floating point values in a robust consistent way.

I've used bitwise operations plenty in other general purpose or lower level programming languages; things like defining compact data structures, handling binary I/O protocols, file formats, etc. 
But I can't think of a single instance in years of using OpenSCAD where I felt like it needed them.

OpenSCAD is strictly designed for *describing geometry*, and general I/O isn't even part of the language.  I honestly am having trouble just trying to imagine a scenario where bitwise operators would be needed or helpful in describing geometry.

Anyways, in the world of geometry, the Boolean operations `union()` (OR), `intersection()` (AND), `difference()` (AND NOT), are the ultimate "bitwise" ops (well, more like point-wise or voluime-wise) :smiley:

rcolyer, 2020-06-01:
You could do bitwise-and compactly if you needed it for some mathematical purpose:

```
function and(a,b) = (a<1 || b<1) ? 0 : ((a%2>=1) && (b%2>=1) ? 1 : 0) + 2*and(a/2, b/2);
```

And if needed, sequential assignment operations can be done in functions using a chain of calculations with the let() feature for variable assignments.  But for most of the things you would use bit-packing for, you can probably do these in a way that better fits the language design by using comprehensions to operate on boolean lists.  Almost every design I've seen created in the language, even with manually creating polyhedrons mathematically, is only computationally heavy at the geometry rendering step, so not much effort has been put into optimizing the computational side.  Even if you put primitives in to convert to integers and do cpu-level bitwise operators, it would still be going through a tree of atomic-guarded shared pointers just to get to each bitwise operator.  I would encourage trying it with functions operating on lists and making use of comprehensions.  If a calculation is so intense that it actually requires native-level performance, then it should be done externally and output some pre-computed source code for an include statement in OpenSCAD to bring into a polyhedron call.

samyk, 2020-06-01:
This is my first OpenSCAD project (I typically use Fusion360 for CAD, but at first blush it appeared to make more sense to use CAD since I needed something more programatic) and I I've ran into a number of features that would simplify this listed below, including missing bitwise operators like >>, <<, and &.

I'm producing (2D) geometry that encodes multiple consumer IR remote control commands into a single "wheel" which will be laser cut. Different commands are at different distances from the center. This is to create a mechanical IR remote control that spins around an IR LED (adjustable on the X axis), with another wheel behind it to produce the carrier wave (normally 38kHz, but most IR receivers will read at much lower rates like 10kHz) that the command signal will ride on top, and the wheels gets spun together either by a small motor or your hand.

Most IR commands (for TVs, receivers, etc) are described by a single number, and the data is encoded in those commands...which would use bitwise operators to easily extract the data and thus produce the geometry, in this case, rectangles of differing lengths around an axis.

OpenSCAD is great but simply may not be the right tool, but I'm contemplating if it's worth me investing some effort into implementing some features that made this project more difficult than I imagined (Fusion 360 allows Python and C but the primitives aren't as simple as OpenSCAD which is why I thought it made more sense).

The specific areas that have made this project quite a bit more challenging:
- no bitwise AND
- no bitwise <<
- no bitwise >>
- no multi-expression return functions (requiring somewhat terse, recursive, single-expression functions and hard-to-follow inline data manipulation)
- no conditional loops (requiring recursive functions)
- limited data types (I can't actually fit the entire command in the numeric data type)
- limited data expressions (I can't store the typical hex value 0x4BB640BF which is how virtually all technical software/documentation would describe a remote command)

While I can see this can all be done in other ways, it's going to be much more complicated for users who are already used to programming in any modern language, which might be acceptable for this project, that's essentially my ask. Is this meant to be an opinionated language that would balk at new features like the above, or is the project interested in a larger user base by making some things easier such as providing familiar programming functionality? Had I not been stubborn and curious about the mental exercise of getting this working, I would have switched back to Processing/Fusion to perform this.

t-paul, 2020-06-01:
Extending the language is certainly on the list, this happens all the time. There's quite a number of discussions open in that regard (e.g. #3087,  #3115), so I'm not sure what specifically you are asking.

In general, language changes need to be discussed for compatibility and future extensibility. So working together towards additional features is great. What's not going to work is dropping a patch for a specific case and assume someone else will take care of the details.

samyk, 2020-06-01:
Thanks @t-paul, that's helpful information! I'll review those tickets to get an idea of the process.

nophead, 2020-06-01:
Also, although not part of the language all those things can be implemented
in user code. For example it would be possible to store long hex values in
strings and loop through them.

Bit-wise AND could be done with a recursive function. The shifts are just
multiple and divide by powers of two, etc. You would be limited to the
range of integers that can be stored as floats.

As pointed out previously, you can have multi-line expressions in function
bodies using let().

Making physical objects from hex strings does seem to be a bit of niche use
case, which is why OpenSCAD doesn't have support for it.

t-paul, 2020-06-01:
Agreed, more flexible data types and maybe even the multi-expressions (without let) would likely be much more useful for general OpenSCAD usage than bitwise operations.

So basically the range is between just adding every feature imaginable and trying to minimize the core (while empowering users to write libraries). Considering the development hours available the tendency is certainly towards a more maintainable core and adding high level features that are not feasible via user code.

But at any rate, it's useful to hear all cases and suggestions so they can be discussed and justified. Maybe it turns out there's a different core change possible making the feature easier to implement in user space.

thehans, 2020-06-01:
rcolyer gave a good example of another way to do Bitwise-AND.

For some of the other operations:  
 - logical shift operations are just multiplying by powers of 2, which is pretty straightforward to create a user-defined function.
 - hexadecimal number literals aren't currently a builtin feature, but can be parsed from strings with a little work.

Here is my own example code for how I would handle most of what you are asking about:
```
// Logical shift operations expect positive integer inputs
function shl(x,y) = x * pow(2,y);
function shr(x,y) = floor(x * pow(2,-y)); 

// convert hex character to number
function hexDigit(c) = 
  let(o = ord(c))
  c >= "0" && c <= "9" ? o - ord("0") :
    c >= "a" && c <= "f" ? o - ord("a") + 10 :
      c >= "A" && c <= "F" ? o - ord("A") + 10 : undef;

// convert hex string to number
function hexToNum(s, i=0, temp=0) = 
  i==0 && s[0]=="0" && s[1]=="x" ? hexToNum(s,2,0) :
    i>=len(s) ? temp : hexToNum(s,i+1, temp*16+hexDigit(s[i]));

// Convert positive integer to list of bits **from least significant to most significant**
// For results going in the other direction, use the following "reverse" function
function numToBits(x) = x==0 ? [0] : [for(n=x; n>=1; n=shr(n,1)) n%2];
function reverse(v) = [for(i=[len(v)-1:-1:0]) v[i] ];

// A few usage examples:
echo([for(i=[0:1:16]) shl(1,i)]);
echo([for(i=[0:1:16]) shr(65536,i)]);

echo(hexToNum("0x64"));
echo(hexToNum("ff"));
echo(hexToNum("0xFF"));
echo(hexToNum("100"));

echo(numToBits(hexToNum("0xF0")));
echo(numToBits(100));
echo(reverse(numToBits(100)));
```

uDude, 2021-10-04:
LOL, the pooh poohing of the indicated features.  I've had to deal with exactly what the author is talking about and resorted to scripting (ohh those evil low level scripting languages) to produce code.

Right now I have a geometry that has an item on one end, the other end, both ends, or none.  So I did what any experienced scripter would do:

if ( position & RIGHT )
  geometry

if  position & LEFT
  mirror geometry

Look if 1 = RIGHT and 2 = LEFT, then BOTH is automatically included.  Alas I write additional code to get around missing bitwise features.

Same as the author, I use external scripts to generate OpenSCAD code to promote strings to object names ---  an "eval"ish thing would probably cover it.

Leave closed.  We all have to create the same libs so maybe they should be considered as adjuncts mentioned on the site.

I worked at an office where this was occurring across a massive project and several people invented their own versions which later had to be consolidated.

jordanbrown0, 2021-10-05:
Implementing bit-wise arithmetic is obviously easy, but it seems like bitwise arithmetic here is really just a poor substitute for what you're really doing, which is set arithmetic.  ("position" is a set with two possible members "left" and "right", and you are asking questions like "is left in position?" and "are either left or right in position?".  "Both" is a constant set with left and right in it, and "does it have either?" is "is the intersection of Both and position non-empty?".

I don't know what the details would be, but I'd look towards set arithmetic (perhaps as a variation on object support) rather than bitwise arithmetic.

In particular, I have a hard time coming up with good use cases for the shift operators.

rewolff, 2023-11-14:
FYI: I have a use case for bitwise AND that does not easily translate to sets or other workarounds. 

I don't like arranging my objects in the slicer, I want their relative positions "reproducible", so I do that in openscad. In my project I have a variable that triggers: "make the build-pate". 

In the project that triggered me coming across this: I have split the whole project into 8 pieces because the printer can't handle anything bigger. I can arrange everything on one build-plate (it fits) but that risks the print failing at 90% and having to restart the whole thing. It has happened twice already on this project. (for smaller-than-everything at once "batches"). 

Anyway, I have a smaller sub-plate that prints in about a day. but during the past weeks I've had to reprint a subset every now and then. I would like to have an "objects to print" bitvector that selects which I'd like to print, including the support infrastructure. For the currently printing object that was part of a plate-of-five, I disabled some "support" for this one by accident, thinking it was for one of the other four. Having a bitvector of "stuff to print" I can then make the support for each of the subobjects conditional on the same bit of the "stuff to print" variable.

gsohler, 2023-11-14:
i miss that too, i will try to make a PR(hopefully i dont have to touch the
lexer ...)

rewolff, 2023-11-14:
Just another "general" remark. 
With the openscad community a question of this nature for an enhancement is often met with: Oh, but what you specifically want to do can be done differently. 

As a workaround that works. And if it is a very "single use case" kind of feature, it makes sense to hold off catering to everybody's wishes. 

The workaround can help the original submitter finish his project  in a couple of days, but a generically useful feature like this... should IMHO be kept open as a "feature that might be implemented once the right person notices it". Last time I submitted such a request my "time to implement" estimate was 10-20 minutes (for someone familiar with the code). After spending an hour in discussions about why the workarounds don't work for everybody, I decided to try to tackle it myself and ended up "on budget" after some having to familiarize myself with the code.

rewolff, 2023-11-14:
On an implementation note: if you make ```rotate (10 AND 12)  ... // rotate by 8 degrees ``` work, that's when you will have to touch the lexer. If you just make an AND function ``` rotate (and (10,12)) ... // rotate by 8 degrees``` then, I think you can get away with "not touching"...

nophead, 2023-11-14:
Bitwise operators work on integers but OpenSCAD only has floats which can only represent a limited number of bits of integer. You could use vectors of bools and some functions for a more general solution.

gsohler, 2023-11-14:
its working

BTW: its not a big hassle to convert a double to an int when used with its
only ...

https://imgur.com/a/C17ByuN

rewolff, 2023-11-14:
On Tue, Nov 14, 2023 at 04:01:12AM -0800, Chris wrote:
> Bitwise operators work on integers but OpenSCAD only has floats
> which can only represent a limited number of bits of integer. You
> could use vectors of bools and some functions for a more general
> solution.

I hope that the floats used are DOUBLEs. Those can represent integers
up to 2^32 accurately. Otherwise there is a limitation of this only
working on 24 bits.

jordanbrown0, 2023-11-15:
Indeed, doubles are good for something like 53 bits.

It sure seems like a set or array would be a good match for this problem.

An input like `[0,1,0,0,1,0,0,1]` seems comparably inconvenient to 0x49.  Better in some ways, worse in others.

Bitmasks seem like such a 1970s solution to the problem.

But I guess that if somebody really wants to implement `and(a,b)`, `or(a,b)`, and `not(b)`, or even `&`, `|`, and `~` operators, yeah, whatever.  They're language clutter, but it's not like there isn't other language clutter.

jordanbrown0, 2023-11-15:
Adding functions does not require touching either the parser or the lexer.  Adding operators requires touching the parser.  I don't think adding operators would require touching the lexer, but I'm not sure.

jordanbrown0, 2023-11-15:
Oh, and shift functions or operators.

jordanbrown0, 2023-11-17:
See PR #4833.  I'm not sure it's a good idea - in fact, I think it's rarely the right answer - but it's simple enough.

rewolff, 2023-11-17:
On Thu, Nov 16, 2023 at 11:18:49PM -0800, Jordan Brown wrote:

> See PR #4833.  I'm not sure it's a good idea - in fact, I think it's
> rarely the right answer - but it's simple enough.

Jordan, Good work! Thanks! 

I think there is atypo. You write 2^32-1 where you mean 2^31-1. 

Hmm. Oh! ... OK. Now I understand. 

Maybe just say: unsigned integers from 0 to 2^32-1 work as
expected. Integers outside that range or non-integer values are
currently truncated to unsigned 32 bit integers. This includes
negative numbers. If you know how 2s complement and the truncation
work, numbers outside 0 .. 2^32-1 could be used, but are unsupported.

For "future compatibility" I'd just reserve EVERYTHING outside 
the first 32 bits. Do not predict 53 bits.

If it were me to extend this, I would just convert a double to a 64bit
longlong. in the range 2^53 .. 2^63 you're loosing bottom bits. You
cannot OR something with 2^61 and then test for the lower 8 bits (or
so)... Anyway, the official "API" is now 0..2^32-1, outside that
behaviour is currently documented but unsupported (=may change at any
time) and when it is extended, you get 0..2^53-1 and outside that
unsupported, again possibly with documeented behaviour.

Consider supporting "-1" as an "all bits set": a noop with "and". 

	Roger.

-- 
** ***@***.*** ** https://www.BitWizard.nl/ ** +31-15-2049110 **
**    Delftechpark 11 2628 XJ  Delft, The Netherlands.  KVK: 27239233    **
f equals m times a. When your f is steady, and your m is going down
your a is going up.  -- Chris Hadfield about flying up the space shuttle.

jordanbrown0, 2023-11-17:
I started out converting to uint64_t, but binary-not yielded bad results, because with small inputs it yielded numbers in the vicinity of 2^64 and on conversion back to double the low-order bits are lost.

jordanbrown0, 2023-11-17:
It's been pointed out in IRC that the 0xHHHH syntax is a breaking change, because OpenSCAD allows identifiers that start with digits(?!?).  Thus, e.g.,
```
0x0f = 14;
```
is legal today and would be broken by this change.

Whether it is a *good* thing that it's legal today is left as an exercise for the reader.

nophead, 2023-11-17:
It is good for me because it allows real names for parts that I model, which sometimes begin with numbers.

jordanbrown0, 2023-11-17:
Alternative strategies might include a function that accepts a hex string and returns a number, or a Python-style `x'foo'` syntax like `h'7f'`.

jordanbrown0, 2023-11-17:
Note that it's only breaking for identifiers that match the 0xHHHHHHHH pattern; identifiers that don't start with `0x` would not be affected.  Similarly, identifiers that have non-hex characters after the `0x` would not be affected; `0x123z` would continue to be a valid identifier.

jordanbrown0, 2023-12-03:
Let's have comments on the proposed change from #4833 over there; no more comments on that proposal here, please.

amatulic, 2025-01-11:
I'll just chime in that I had a need for bitwise operations today. It's useful for generating surfaces via the marching cubes algorithm (in which 256 cube configurations can be mapped to 15, and one of the 256 configurations is identified by the list indices of the cube vertices greater than the isosurface value OR'd together), as well as FFT (admittedly an unusual use case for OpenSCAD but bitwise indexing makes things more efficient).

gsohler, 2025-01-12:
Implementing the Marching cube algorithm in openscad sounds interesting.
Could you elaborate on your intent/design goal ?

jordanbrown0, 2025-01-12:
> Implementing the Marching cube algorithm in openscad sounds interesting.
> 
> Could you elaborate on your intent/design goal ?

Sounds like a fine discussion for the mailing list, but please not on GitHub.

amatulic, 2025-02-28:
>Implementing the Marching cube algorithm in openscad sounds interesting.
>Could you elaborate on your intent/design goal ?

@gsohler -- as jordanbrown0 requested, I won't elaborate further other than to say the project is done and now live [here](https://github.com/BelfrySCAD/BOSL2/wiki/isosurface.scad). I don't know anything about a mailing list for OpenSCAD, but my email address should be available from my GitHub profile if you want to discuss it.

closed 2025-03-14 by kintel

chrisnorman7, 2025-08-03:
Hi,
Bitwise operators would make my life a lot easier. I'm trying to write braille in OpenSCAD, and being able to use & would mean I could convert a character like ⠓ (the letter h) easily.

remainder = ord("⠓") - 10240
dot1 = remainder & 1;
dot2 = remainder & 2;
dot3 = remainder & 4;
dot4 = remainder & 8;
dot5 = remainder & 16;
dot6 = remainder & 132;

Currently, I'm using Python to code the module for me. That could be because I'm a newbie though.

Thank you for a wonderful bit of software in any case! I love it! Thank you.

bitbasher, 2025-08-04:
if you are able to use the developement snapshot the binary operators are already implemented in it

[documented here ](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Mathematical_Operators#Binary_arithmetic)

commit bb7ee12c (2025-03-14): Bitwise operators, hexadecimal constants.  Fixes #3345. (#4833)