openscad/openscad #3349: Wrong source location identified when picking in 3d view

t-paul, 2020-06-07:
Moving [issue](https://github.com/openscad/openscad/pull/3276#issuecomment-640123459) from #3276.

I did a quick test, and the second example I loaded seems to return the wrong pick result.
In the following small example, any picking done in the negative green surface will indicate that we actually picked the red ring:

[image]

```
difference(){
  union() {
    color("Green")
      translate([25, 0, 0]) 
        cylinder(h=7, r=4);
        
    color("Red")
      translate([8,0,0])
        linear_extrude(4)
          difference() {
            circle(15);
            circle(13);
          }
  }
  translate([25, 0, 2])
    cylinder(h=8, r=3);
}
```

nophead, 2020-06-07:
Does setting the correct convexity in the linear_extrude make any difference? That example doesn't even display properly for me.

Tomatower, 2020-06-07:
I can confirm the issue.

This is what the render-by-id looks like (the red-tone is `index() * 64` for better contrast)

[image]

the negative object has the same `index()` as the big red ring. 

Switching to ThrownTogether rendering lets me select everything fine

I have not yet fully understood how the tree normalization implacts the identifiers. and i guess this is the source of the issue.