openscad/openscad #3353: Trying to render the joiner from BOSL causes CGAL assertion failure.

benjamind2330, 2020-06-08:
You can render in openscad the BOSL half_joiner2, i have even exported to stl and printed one. But when attempting to combine it with other things, you cannot render it. You also cannot render the joiner at all. The set of joiners are described here: https://github.com/revarbat/BOSL/wiki/joiners.scad#half_joiner2

You get the error:
```
ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation! Expr: e_below != SHalfedge_handle() File: /usr/include/CGAL/Nef_3/SNC_FM_decorator.h Line: 426
```
**Code To Reproduce Bug**
```
include <BOSL/constants.scad>
use <BOSL/transforms.scad>
use <BOSL/joiners.scad>
joiner();
```
Then use F6 to render.

**Expected behavior**
I expect to be able to render this simple combination of joiner and shape.

**Additional context**
OpenSCAD Version: 2019.05

thehans, 2020-06-08:
Looks like related to #2991

Quick workaround:
If you run F6 without ever doing F5 then its fine.
Alternatively you can use "Design -> Flush caches" if F5 has been run before F6.

I've been working on some other features lately, but I may try to finally tackle this issue soon.

benjamind2330, 2020-06-09:
yep that was it, I will close this as duplicate.

closed 2020-06-09 by benjamind2330