openscad/openscad #3354: Incorrect geometry after DXF import

olemis, 2020-06-12:
DXF not rendered correctly

**OpenSCAD version** : 2019.05
**Input file**: Octanorm [S-108](https://www.octanorm.com/us/products/product?text=S%20108&id=2101&pbc=1-1-125-500-900-2101) profile section upright (classic system).
**Result with online DXF viewer**

[image]

**Result with OpenSCAD**

[image]

[image]

t-paul, 2020-06-12:
That file can be downloaded only with login. Please put that somewhere accessible if allowed by license or provide a different test case.

olemis, 2020-06-12:
@t-paul please check this [alternate download location](https://github.com/cubanmakers/3DPrintModels/blob/master/assets/extrusions/octanorm/official/files/cad/S%20108.dxf).

olemis, 2020-06-12:
The same issue happens with all extrusion sections (i.e. `.dxf` files) in that folder . Attached picture shows the result of rendering them all after clipping imported volume to expected extrusion width and height. I did so in order to be able to render them altogether without interference of incorrect rectangular area.

[image]

 Feel free to try them all as test input once a patch be submitted . I'll a few more soon .

p.s. once this will be fixed (in `master`?) are there `openscad` nightly builds (or equivalent) available for download ?

t-paul, 2020-06-12:
I'm getting a number of warnings. I guess one of those unsupported objects leave behind some extra data which is coming though as that big rectangle.
```
WARNING: Unsupported DXF Entity 'SEQEND' (1) in "../../tmp/S 108.dxf". 
WARNING: Unsupported DXF Entity 'ATTRIB' (f) in "../../tmp/S 108.dxf". 
WARNING: Unsupported DXF Entity 'VIEWPORT' (1) in "../../tmp/S 108.dxf". 
WARNING: Unsupported DXF Entity 'ATTDEF' (f) in "../../tmp/S 108.dxf". 
WARNING: Unsupported DXF Entity 'TEXT' (17) in "../../tmp/S 108.dxf". 
```
Snapshot builds should show up automatically on the [download page](https://www.openscad.org/downloads.html#snapshots) (MacOS only every 2 days as we don't have enough server time to build more often).

olemis, 2020-06-12:
> I guess one of those unsupported objects leave behind some extra data

🤔 ... understood . Could you please point me out to the DXF library source code?

olemis, 2020-06-12:
> Could you please point me out to the DXF library source code?

@t-paul is [dxfdata.cc](https://github.com/openscad/openscad/blob/4ef45b005c0d3c4fa2b6b375652c5bf9ef051df4/src/dxfdata.cc) the right file ?

t-paul, 2020-06-12:
Currently it's using an old [handcrafted parser](https://github.com/openscad/openscad/blob/master/src/dxfdata.cc) with limited compatibility.

The plan was to move to the BRL-CAD parser but that project is not quite finished (current state in https://github.com/openscad/openscad/issues/3044)

olemis, 2020-06-12:
For the moment, I found a way to work around this by exporting DXF to SVG using [QCAD command line tools](https://qcad.org/en/products/qcad-command-line-tools) .

t-paul, 2020-06-13:
That extra geometry is actually the page outline which gets messed up by the importer, but the actual geometry you want is layer `Profil`:
```
import("S 108.dxf", layer = "Profil");
```

closed 2020-06-13 by t-paul