openscad/openscad #3355: Faces missing on `polyhedron` when adding a `cube` on render, but not on preview

TooAngel, 2020-06-12:
Having a simple `polyhedron`:
```
polyhedron(
    points= [
        [0, 0, 0], [10, 0, 0], [10, 10, 0], [0, 10, 0],
        [0, 0, 5], [10, 0, 5], [10, 10, 5], [0, 10, 5],
    ],
    faces = [
                [0, 1, 2, 3], [4, 5, 6, 7],
                [0, 1, 5, 4], [1, 2, 6, 5], [2, 3, 7, 6], [3, 0, 4, 7], // These are missing on render after adding a cube
            ]
);
```
(looks good)

Adding a `cube`:
```
cube([1, 1, 1]);
```
The outer faces are missing. Translating the cube out of the `polyhedron` does not help.
In the preview, it looks good, just on render.
The `Console` only shows `UI-WARNING: Object may not be a valid 2-manifold and may need repair!` and no error.

---

I discovered the issue with a slightly more complex `polyhedron` and adding a `cube`, the `Console` shows `ERROR: Alternate construction failed. CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation! Expr: e->incident_sface() != SFace_const_handle() File: /usr/include/CGAL/Nef_S2/SM_const_decorator.h Line: 326`

Issues which might be related:
#2842 or #2412

t-paul, 2020-06-12:
The polyhedron has most of the faces flipped (facing inward) which can be seen in ThrownTogether view (F12).

See also the [FAQ](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#Why_is_my_model_appearing_with_F5_but_not_F6?) entry for more links.

closed 2020-06-12 by t-paul

TooAngel, 2020-06-12:
Sorry, missed that somehow, thanks

t-paul, 2020-06-12:
No problem, we need to find a more obvious way to highlight those issues, ideally in the normal preview mode.