openscad/openscad #3356: Crash when rendering (F6)

DisDis, 2020-06-13:
OS: Ubuntu 19.10 64bit
Openscad 2019.05

Chash when rendering F6

reproduce file: https://github.com/DisDis/dslideshow/blob/4d6587bcdbe0f468278ad2e79303ef6b524e7d54/doc/photoframe_back_cover.scad

t-paul, 2020-06-13:
I can't reproduce the crash, but there's include files missing.

thehans, 2020-06-13:
I was able to reproduce, but the example requires downloading the rest of the repo, from which that link is just one of the files.

The issue appears to be with your included pcb models, which if rendered individually, give non-manifold warnings.  I think the reason they do is that the "blocks" on the boards are sharing the exact Z height as the face of the PCB, which can cause problems sometimes.  I would suggest offsetting those to slightly intersect with the board, which would hopefully workaround CGAL's strictness in this case.

And I'm not sure your goal, but typically in a model like this, the PCBs are only desired during F5 Preview as a mockup, and during render you would just want the case model.
You can achieve this by wrapping the last few lines of that file with `if($preview) { ... }`

Like this:
```
if ($preview) {
//RasPi4
translate([pcbRPi4X, pcbRPi4Y, pcbRPi4Z]) rotate(a=[0,0,-90]) raspberry4();

//Display Driver 
translate([(width-pcbDriverW)/2+pcbDriverW/2,(height-pcbDriverH)/2+pcbDriverH/2,back_thickness+10]) rotate(a=[180,0,-90]) translate([-pcbDriverW/2,-pcbDriverH/2,0])  pcbDisplayDriver();

//PoweDistib
translate([pcbPowerX, pcbPowerY, pcbPowerZ]) pcbPowerDistribution();
}
```

DisDis, 2020-06-13:
@thehans 
Thank you, I have a little experience with OpenScad.

> if ($preview) {

This fixes the crash.

How can I help with the problem? Or should I close issue because there is a workaround.
@ t-paul , you should clone the repository because photoframe_back_cover.scad needs other files.

t-paul, 2020-06-13:
It would still be interesting to find the reason for the crash. There's some chance it's caused by #2847.

DisDis, 2020-06-13:
@t-paul How can I get data(log and etc.) to verify this guess?

kintel, 2024-03-31:
Status as of 2024-03-24: Doesn't crash any longer, but doesn't produce an entirely correct geometry with neither CGAL or Manifold.

pca006132, 2024-09-05:
Is this correct? I am testing #5282

[image]

kintel, 2024-09-19:
Seems to work well with Manifold. Also no longer crashes with CGAL.
Closing this.

closed 2024-09-19 by kintel