openscad/openscad #3359: bountysource 2-year limit

dorkmo, 2020-06-17:
need to review old bounties, just got this email

Hi, You are receiving this email because we are updating the Bountysource Terms of Service, effective 1st July 2020.What's changing?We have added a Time-Out clause to the Bounties section of the agreement:2.13 Bounty Time-Out. If no Solution is accepted within two years after a Bounty is posted, then the Bounty will be withdrawn and the amount posted for the Bounty will be retained by Bountysource. For Bounties posted before June 30, 2018, the Backer may redeploy their Bounty to a new Issue by contacting support@bountysource.com before July 1, 2020. If the Backer does not redeploy their Bounty by the deadline, the Bounty will be withdrawn and the amount posted for the Bounty will be retained by Bountysource.You can read the full Terms of Service hereWhat do I need to do?If you agree to the new terms, you don't have to do anything.  If you have a bounty posted prior to June 30, 2018 that is not currently being solved, email us at support@bountysource.com to redeploy your bounty.  Or, if you do not agree with the new terms, please discontinue using Bountysource.Thanks for readingBountysource Team
--

t-paul, 2020-06-17:
Thanks, that was just mentioned in #391. Dedicated issue is helpful. That's a very short deadline they give here :disappointed:.

Lets see what the reaction is. For me that sounds like BountySource is dead.

t-paul, 2020-06-17:
Ok, sent out an email politely asking to 1) extend the deadline and 2) dropping that idea completely.

MichaelAtOz, 2020-06-17:
I also asked how the two year will be processed into the future, ie daily check or start/end-of-month etc. [noting I have not had a reply from 'support' on past problems...]
Most bounties are > 2 years, so we should 'claim' them & re-donate.

TLC123, 2020-06-17:
Yeap  'claim' them & re-donate in a meaningful way sounds fair to me. Better than being ripped of by B.S.

t-paul, 2020-06-17:
Reference: https://github.com/bountysource/core/pull/1480

t-paul, 2020-06-17:
ToS change reverted for now: https://github.com/bountysource/core/pull/1498 but comment on the other issue says:
>  ToS will be revised and clarified and in the future

I guess we have to wait and see what's going to happen. Maybe still prepare to move out.

ioquatix, 2020-06-18:
Have you considered adding organisation funding via GitHub sponsors? I'd definitely buy everyone a coffee.

kintel, 2020-06-18:
We're on the waiting list for GitHub sponsors

MichaelAtOz, 2020-06-18:
[Fees](https://www.bountysource.com/fees), *currently* they take 10% when money is withdrawn. So 'claiming' issues, then doing a 'team contribution' does not lose any value. It will lose the original sponsor connection, ie 'approving' the claim, so we would need to canvas those contributors.

> We Backed
$100 active bounties
$714 paid bounties
1 active bounty
25 bounty hunters
$2,350 total bounties on our issues ($1060 on multi-thread, so $1290 on others)

I'm trying to get a consolidated list of open bounties with dates, doesn't seem to exist, I've asked support. It appears there are 25.

t-paul, 2020-10-05:
Closing this, hopefully for good.

closed 2020-10-05 by t-paul