openscad/openscad #3360: CGAL error in CGALUtils::applyBinaryOperator difference

andysc, 2020-06-27:
Parsing design (AST generation)...
Compiling design (CSG Tree generation)...
Rendering Polygon Mesh using CGAL...
ERROR: CGAL error in CGALUtils::applyBinaryOperator difference: CGAL ERROR: assertion violation! Expr: itl != it->second.end() File: /Users/kintel/code/OpenSCAD/libraries/install/include/CGAL/Nef_3/SNC_external_structure.h Line: 1153

My model has 4 imported STLs (generated elsewhere), joined together into a square

The problem seems to be data-dependent, as I don't get it with some images, but do with most of them.
Simplifying the model helps - I removed the "lid" of the box and it worked, but then I changed the images and it broke again.

OpenSCAD `2019.05` on Mac Catalina.

MichaelAtOz, 2020-06-27:
CGAL Assertion violations are 99.9% the result of a non-manifold or self-interseting STL (or a bad polygon/hedron).

You need at least two objects to cause CGAL to be called.
So do 
```
import("your.stl");
cube(1);
```
That forces a `union()`, it will tell you which STL is bad.

andysc, 2020-06-27:
Thanks - yes, it's the imported STL... a lithophane image.
>UI-WARNING: Object may not be a valid 2-manifold and may need repair!
I guess I should get a clean one of those first.
Thanks for the pointer :)

andysc, 2020-06-27:
meshmixer fixed it in an instant!
Thanks again :)

andysc, 2020-06-27:
ah, I spoke too soon. It looked OK in preview, but when I rendered it, the lithophane STL disappeared.
No errors this time, though... 
```
Total rendering time: 0 hours, 0 minutes, 0 seconds
Top level object is a 3D object:
Simple: yes
Vertices: 8
Halfedges: 24
Edges: 12
Halffacets: 12
Facets: 6
Volumes: 2
Rendering finished.
```
This was the import and cube(1) test.
It gave me a cube after the render

nophead, 2020-06-27:
The error will only be shown once. Try Flush Caches to see it again.

andysc, 2020-06-27:
ah, so it does!
```
ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation! Expr: e_below != SHalfedge_handle() File: /Users/kintel/code/OpenSCAD/libraries/install/include/CGAL/Nef_3/SNC_FM_decorator.h Line: 427
```
thanks - back to fixing these STLs up, then.

closed 2020-06-27 by MichaelAtOz