openscad/openscad #3361: Creating objects composed that include future subtractions

raffaeler, 2020-06-28:
I would like to create a library of objects that are composed by:
- parts that need to **merged** (union) with the main drawing
- parts that are for helpful only during **design**: rendered for convenience but never exported in the final drawing
- parts that need to be **subtracted** (difference) from the main drawing

Example:
- main object: a box
- library object: an arduino board composed by:
  * 4 pcb supports to hold the arduino
  * the pcb board is for design only: useful to understand how the pcb is oriented and how big is the board to reserve enough space
  * 1 usb receptacle that must be subtracted from the lateral box to create the hole

Managing those 3 categories of parts separately is difficult because the 3D position of one sub-part is tied to the others.

Is there any way to express this?
Maybe tagging or naming the sub-parts could help.

If this is already possible, please point me to the right direction.
TIA

nophead, 2020-06-28:
I do exactly that sort of thing with my library [NopSCADLib](https://github.com/nophead/NopSCADlib). I have a generic [printed plastic box](https://github.com/nophead/NopSCADlib#Printed_box) that I can fit around PCBs, etc, add supports and punch out holes. I do that by having a blank case that is customised by passing it a list of dimensions, screw type, etc. It also takes three children objects. The first is internal additions, the second is holes and the third external additions.

It can make boxes as different as these two:
This houses an arduino, transformer, LCD display, etc.

[image]

[image]

And this houses an RPI0, and environmon+ board and a fan.

[image]

I also have customisable boxes made from CNC cut panels and printed plastic parts. In those cases I again pass a list of the inner dimensions which are calculated from add the sizes of the parts inside with the required spacing. It then provides blank panel templates that I subtract all the holes from in 2D.

[image]

raffaeler, 2020-06-28:
Interesting, but I can't really understand how you can define a single part (arduino or whatever else) that defines sub-parts being either added to the main drawing, just displayed but not added, or subtracted.

If I understand well your strategy, you defined a box accepting the parts to be drawn and subtracted and I have no doubts that it work.
But those subparts have to be defined separately, right?
Or I may have not understood how you did it :)

nophead, 2020-06-28:
The Arduino is an instance of a pcb in my library. It is just a list of properties like this.
```
ArduinoUno3 = ["ArduinoUno3", "Arduino Uno R3", 68.58, 53.34, 1.6, 0, 3.3, 0, "#2140BE", false, [[15.24, 50.8],[66.04, 35.56],[66.04, 7.62],[13.97, 2.54]],
    [[30.226, -2.54, 0, "2p54socket", 10, 1],
     [54.61,  -2.54, 0, "2p54socket", 8, 1],
     [36.83,   2.54, 0, "2p54socket", 8, 1],
     [57.15,   2.54, 0, "2p54socket", 6, 1],
     [64.91,  27.89, 0, "2p54header", 2, 3],
     [18.796, -7.00, 0, "2p54header", 3, 2],
     [ 6.5,   -3.5,  0, "button_6mm"],
     [4.7625,  7.62, 180,"barrel_jack"],
     [1.5875, 37.78, 180,"usb_B"],
     [46.99,  17,    270,"pdip", 28, "ATMEGA328", true],
    ],
    [],[],
    inch([
     [-1.35, -1.05],
     [-1.35,  1.05],
     [ 1.19,  1.05],
     [ 1.25,  0.99],
     [ 1.25,  0.54],
     [ 1.35,  0.44],
     [ 1.35, -0.85],
     [ 1.25, -0.95],
     [ 1.25, -1.05],
    ]),
    M2p5_pan_screw
   ];
```

 I can draw it with ```pcb(ArduinoUno3)``` but I can also get its dimensions to place it with functions ```pcb_length(ArduinoUno3)```, ```pcb_width(ArduinoUno3```), ```pcb_thickness(ArduinoUno3)```.

To place the screws, drill holes for them or place bosses I have a module ```
pcb_screw_positions(ArduinoUno3)``` that positions children at the mounting hole positions.

So when I call the module to make the box it looks like this.
```
box = pbox(name = "case", wall = case_wall, top_t = top_thickness, base_t = base_thickness, foot = foot,  radius = case_inner_rad, size = [case_width, case_depth, case_height]);

module case_stl() pbox(box)      { case_additions(); case_holes(); }
module case_base_stl() pbox_base(box) { base_additions(); base_holes(); }
```

The bosses to hold the inserts to mount the Arduino are defined line this:

```
module base_additions() {
    translate([arduino_x, -arduino_y])
        pcb_screw_positions(arduino) {
            insert_boss(arduino_insert, case_height + top_thickness + base_thickness - arduino_z);

            insert_boss(arduino_insert, case_height + top_thickness + base_thickness - arduino_z - 2, 2);
        }
}
```

So the box is generic in the library but is customised by properties and modules in the project. Every object in my library is defined by a named list of properties, every screw, nut and washer, etc. These lists are passed by name to all the functions and modules that deal with the object. either to draw it, or make holes for it, etc.

When I draw the sub assembly diagrams it looks like this.
```
module arduino_assembly() {
    pcb(arduino);

    pcb_screw_positions(arduino)
        translate_z(pcb_thickness(arduino))
            screw(arduino_screw, arduino_screw_length);

    explode(50)
        translate_z(expansion_z - arduino_z)
            pcb(expansion);
}

module base_inserts_assembly(alpha = 1)
assembly("base_inserts") {
    translate([arduino_x, -arduino_y, case_height + top_thickness + base_thickness - arduino_z + eps])
        pcb_screw_positions(arduino)
            insert(arduino_insert);

    stl_colour(pp2_colour, alpha) render() case_base_stl();
}

module base_assembly(alpha = 1)
assembly("base") {
    translate([arduino_x, -arduino_y, case_height + top_thickness + base_thickness - arduino_z + eps])
        arduino_assembly();

    base_inserts_assembly(alpha = alpha);
}
```
And looks like this:

[image]

[image]

nophead, 2020-06-28:
I put the complete project file here so you can see exactly how it is made. The rest of the code is in NopSCADlib that I linked to before.

https://gist.github.com/nophead/d148f2ec63454c9e8e36a14b3e4ccb69

raffaeler, 2020-06-28:
I am new to this world, sorry if I don't get all the details.
Wouldn't it be nice to have a built-in support in openscad to define sub-parts as I asked? Your workaround works for sure, but looks very complex to me.

My goal is just 3d printing, not creating assembly documents. In addition to the arduinos I have my own custom pcbs and several sensors.

In your sources, how do you tell openscad to just print the separate parts (only those that must be printed)?
For example:
- the cover must be exported on the z=0 plane so that the 3d printer can print it
- the tools as well as the arduinos should not be drawn at all

nophead, 2020-06-28:
All the modules that generate STLs to be printed are called ```somename_stl()``` and call module ```stl("somename") ```. That adds them to the BOM by creating an echo file when $bom=2 is passed on the command line.

All STL modules generate the part oriented as it is printed. To make the assembly view I rotate them and translate them into the orientation they are used in.

My framework runs through each STL it sees on the BOM, finds the module called ```somename_stl()``` and calls it to generate the  STL via the command line. To make an assembly view it calls ```somename_assembly()``` and makes a png file.

For a simple project that only has one print bed of STLs I draw the assembly with $preview is true, i.e. when F5 is pressed and I lay out the parts for printing when it is false, i.e. F6 is pressed. 

In that project I actually have this at the end of the file.

```
if($preview)
    main_assembly();
else rotate(90)
    if(1) {
        case_stl();

        translate([lcd_x, lcd_y])
            fence_stl();
    }
    else base_stl();
```

So I could export the STLs with F7 but I normally just have it there to view them and just run my script stls.py to export all of them.

raffaeler, 2020-06-28:
You did a great/huge work, but frankly I find it very complex.

I believe that certain facilities should be part of openscad for multiple reasons:
- to reduce the amount of code needed to reuse a component in a library
- to make the libraries ecosystem grow
- to ease reuse of the libraries. My libraries should be now re-modeled to fit in your framework, which is not worth the pain for a community project
- finally, I also see a license issue. Even when I share code that targets my community I decided to stay away from (L)GPL libraries at all.

Don't get me wrong, you did a great work, but I believe openscad should improve supporting this scenario.
Thanks

nophead, 2020-06-28:
I also have platters.py that allows me to aggregate STLs for printing and panels.py that aggregates DXFs. They run through scad files in the platters and panels directories, which import STLs and DXFs and position them to be printed or routed together.

They look like this (from a bigger project):
platters/spacers.scad
```
include <NopSCADlib/core.scad>

for(x = [0 : 3]) {
    for(y = [0 : 1])
        translate([x * 10, y * 10])
            use_stl("pcb_spacer2560");

    translate([x * 10, -10])
        use_stl("pcb_spacer2560_2");

}

translate([-15, 0])
    use_stl("spacer");

rotate(90)
    use_stl("pulley_part_flange");
```

use_stl() imports the stl and also removes it from the list of parts to be printed and adds the aggregate, spacers.stl, instead.

[image]

nophead, 2020-06-28:
Well I am note sure what you asking to be added to OpenSCAD. The language already supports doing what you want as I have demonstrated. $preview is the main tool for getting an assembly view when you hit F5 and just STLs for printing when you hit F6.

The way to have objects in a library that can both draw themselves, or make cutouts in a box , or position fasteners is to make modules to do each of those things from the same set of parameters. To avoid passing a long list of parameters to each function I put them in a list. If the item is a one off that doesn't need to be parametric I dispense of the list but still have separate modules with a common name prefix to draw the object or make a hole for it. For example:

```
function jack_4mm_hole_radius() = 8/2; //! Panel hole radius for 4mm jack

module jack_4mm(colour, thickness, display_colour = false) { //! Draw a 4mm jack socket with nut positioned for specified panel thickness
```

[image]

Other people use a single module and pass a parameter to change whether it draws itself or its holes but I don't personally like that. I prefer to have separate modules with a common name prefix.

To position parts relative to each other I define them with the most convenient origin and have functions to get the size to allow be to say position a panel meter on a panel. To define things like screw positions relative to a PCB made a module that positions its children on the screw positions. Use it to make the holes in the PCB and position the fasteners.

Other people have made libraries that have the concept of connectors to positions things relative to each other but I don't use that myself.

What else do you need to achieve your goal? I wasn't advocating you use my library. I was using it as an example of how it can be done with OpenSCAD.

nophead, 2020-06-28:
Also you can make modules that both draw an item and position children relative to it. For example my washer module translates any child passed on to its top surface, so ```washer(M3_washer) screw(M_cap_screw, 10)``` will place a screw on top of the washer on top of a surface.

jordanbrown0, 2023-12-13:
I believe this is a duplicate of #921.  Take a look and tell me if you think so.