openscad/openscad #3362: Feature Request: axis scales are correct regardless of orientation

robertlugg, 2020-06-28:
Depending on the view, the numbers on the primary axes are flipped.

This is what I consider to be a good case:
[image]

The next two are the bad cases.  As the view changes the numbers are shown upside down.

Note that 20 is shown as 02 in one of the axis:
[image]

Note that 20s are shown as 50s:
[image]

I actually can't say if this is wrong or how other tools work, but I'm wondering if the numbers somehow adjust so that they are always front facing and flipped to be correct from the viewport.

haarp, 2021-07-29:
Yes, the axes are often not very readable. On top of them becoming rotated upside down, they can also be rotated to become perpendicular to the viewport, rendering them unreadable. Plus, the numbers are often hard to read due to lack of contrast.

The same goes for the X Y Z marker.

I would propose that the numbers always stay parallel to the viewport. In addition, the numbers (and probably the axes themselves) should dynamically adjust their color. A clever way of doing so is to always show the inversion of whatever color is behind them.

jordanbrown0, 2024-11-23:
Note that current development snapshots use a much nicer font.  While the labels can still be up-side down or reversed, they aren't nearly as visually ambiguous.