openscad/openscad #3367: DBus not accepting actions on startup

Lenbok, 2020-07-05:
I would like to control OpenSCAD from emacs via DBus. I have enabled DBus in Preferences-> Features -> input-driver-dbus and Axes -> DBus.  It appears that DBus is not being initialized correctly.  (To me, it seems as though it is only able to execute commands after the user has manually invoked some actions from the menu.)

To start with, I have built the (helpful) openscad-dbus-reference app by @MichaelPFrey

To reproduce:
* Start openscad (with a scad file, doesn't seem to matter which)
* Start the dbus reference app (it indicates connected status, good)
* Press "get actions" to populate the list of all available actions (works, good)
* Select an action, such as "viewActionViewAll" or "viewActionHideConsole" and press "send action" to send it

Actual result:
* Nothing happens

Expected result:
* The action should be executed

Now, manually choose some items from the menu (in my case it was View -> Zoom in, Preferences, etc). After enough tinkering, you can try sending the action from the reference app and the action succeeds. From then on, the DBus interface seems to be working (both from the reference app, and from my test functions within emacs).

I am on Ubuntu 20.04 using openscad-nightly 2020.06.14.nightly (git be16bd8) but get the same behaviour with release 2019.05

t-paul, 2020-07-05:
Is the application having input focus? For normal input devices, events are only forwarded to the active window. This may need some special handling for DBus.

Lenbok, 2020-07-05:
Once DBus is up and running, I'm happily sending commands from my emacs (or the reference app), which are the ones that have focus, so that is probably not the issue.

t-paul, 2020-07-05:
Strange, I'm not aware of other special cases for the action list vs. other calls, but I guess that's what happens with bugs sometimes :wink:.

Lenbok, 2020-07-05:
It's not just the action list (those are just the ones I've been trying out first). The other commands for zooming, etc in the reference app also do nothing.

Lenbok, 2020-07-26:
Once the DBus interface is up and running, it's really handy. 

I've put together a simple emacs package utilizing it, at https://github.com/Lenbok/scad-dbus

yoshinari-nomura, 2026-02-06:
This issue seems to have disappeared in the current version of nightly build: openSCAD-2026.02.04-x86_64.AppImage.
This may have been improved through recent refactoring of the MainWindow constructor's code.

Lenbok, 2026-02-06:
Actually, I just tested this now using the same nightly and it is *not* fixed. Switching back to my build from my PR works. Please reopen. 

(edit: @yoshinari-nomura do you have any of the views other than the 3d view showing? I have all other windows turned off)

Lenbok, 2026-02-06:
@yoshinari-nomura You can use the appimage from https://app.circleci.com/pipelines/github/openscad/openscad/8220/workflows/00a47b16-f51c-4fbe-b3d5-c558c657eb3a/jobs/31369/artifacts which is the most recent build from my PR (although I notice that also some changes on master have resulted in several DBus actions disappearing #6618 )

yoshinari-nomura, 2026-02-06:
@LenBok, thank you, I've tried your AppImage and confirmed it worked as expected with scad-dbus. But #6618 is bad news...