openscad/openscad #3368: Crashes when trying to render STL file on headless system

TheAssassin, 2020-07-06:
I'm trying to render STL files headlessly. However, OpenSCAD just segfaults.

Log:

```
Could not initialize localization.
Compiling design (CSG Products normalization)...
Normalized CSG tree has 15 elements
Unable to open a connection to the X server.
DISPLAY=
Can't create OpenGL OffscreenView. Code: -1.
Geometries in cache: 39
Geometry cache size in bytes: 699496
CGAL Polyhedrons in cache: 0
CGAL cache size in bytes: 0
Total rendering time: 0:00:00.154
<segfault>
```

Command: `openscad -o "$out_file" "$in_file" --viewall`
Version: `OpenSCAD version 2020.06.22` (I've built an AppImage of OpenSCAD recently, I could get a commit hash if needed.)

thehans, 2020-07-06:
What do you mean render STL files?
What is the file extension of your `$out_file`?  
Are you trying to render an png image of an imported STL or are you trying to generate STL output from scad?  
If the outfile is .stl then --viewall doesn't really make sense.  
If you want .png output then I think you need to setup and run `xvfb` to get a offscreen framebuffer.

I don't know the exact commands needed, but I believe the test servers rely on this script, which might help at least as a reference:
https://github.com/openscad/openscad/blob/master/tests/virtualfb.sh

TheAssassin, 2020-07-06:
> What do you mean render STL files?

Erm... `openscad --export out.stl in.scad`, basically.

> Are you trying to render an png image of an imported STL or are you trying to generate STL output from scad?

Of course I'm trying to render STL files from `.scad` files. I use the same method to generate `.png` screenshots (that's where the `--viewall` is used; I use the same bash function for both applications).

> If you want .png output then I think you need to setup and run xvfb to get a offscreen framebuffer.

I know how to work around this. But my point is it shouldn't be needed in the first place, as no window is opened. I haven't had a look at the code, all I can tell is to me as a user it is not obvious why an X server would be needed.

The X server is also needed for `.stl` output, as far as I can see...

If X11 is really needed for `.png` and/or `.stl`, maybe the logs should reason this or maybe even suggest to use Xvfb?

t-paul, 2020-07-06:
Yes, PNG needs X11 connection for the OpenGL context. There's no support for creating EGL offscreen contexts. STL should not require the X11 connection.

So the bug seems to be that PNG crashes when the OpenGL context creation fails. A quick test with `unset DISPLAY` and generating STL works with `master`.

TheAssassin, 2020-07-06:
> A quick test with unset DISPLAY and generating STL works with master.

Just tried again, appears to work.

> So the bug seems to be that PNG crashes when the OpenGL context creation fails. 

```
Unable to open a connection to the X server.
DISPLAY=
Can't create OpenGL OffscreenView. Code: -1.
```

Why doesn't the tool just abort at this point? It continues to run and then just segfaults.

robertlugg, 2020-07-12:
How might this be handled on a Windows build?  Does it also require DISPLAY to be set?

t-paul, 2020-07-12:
Windows uses WGL via https://github.com/openscad/openscad/blob/master/src/OffscreenContextWGL.cc

kintel, 2024-03-31:
We now support rendering using EGL, eliminating the need for X11. Requirement is that OpenCSG is >= 1.6.0.

closed 2024-03-31 by kintel