openscad/openscad #3370: Using special font characters

catmandu13, 2020-07-09:
I'm trying to use a character listed as Unicode Code Point U+1F770.

I've tried using 
"\u1F770"
chr(128880)
"🝰"

and various versions of the same. Hunting around I find that basically anything past "\u1FFE" doesnt seem to register even if it is part of the font set used to print. Is this a OpenSCAD limitation, or is there another method to use the higher level UTF characters?

What's weird is in the code editor I can paste the "🝰" character and it shows, do a preview and nothing but a box, but i assume that's a difference in the GUI and the compiler?

thehans, 2020-07-09:
Please provide a more complete example.  Are you trying to `echo()` this character to console or use it with the `text()` module?

What font are you using in your editor which provides this symbol?  
Is the font free to download, or bundled with a proprietary OS?

If you're using `text()`, have you set its `font` parameter to the same as the editor uses?
If you're using `echo()`, I'm not sure what determines the default console font but I don't think that's currently configurable through OpenSCAD settings.  Could be a nice feature though.

catmandu13, 2020-07-09:
Thanks for the reply. 
I'm using:
text("\u1F770", size=4, valign="center", halign="center", font="Symbola:style=Regular");

"Symbola" being the font that has the symbol.
Available here: https://dn-works.com/wp-content/uploads/2020/UFAS-Fonts/Symbola.zip

Win10 seems to recognize the symbol with it's default library as I can paste it in notepad and it shows it, but only certain fonts will change the style. So i image there's a default it refers to if it's not present in a font, but I couldn't figure out which one. So I sought out one that was known to have it.

t-paul, 2020-07-09:
As shown in the [manual](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Text) you have to use either 4 digit `\u0123` (lower case `u`) or 6 digit version `\U012345` (upper case `U`).

This works on my Linux system:
```
text("\U01F770", size=4, valign="center", halign="center", font="Symbola:style=Regular");
```

[image]

catmandu13, 2020-07-09:
Ah!...good to know about the uppercase U!

Copy and pasting the code provided by t-paul:
On Windows 10, still get a box:
[image]

t-paul, 2020-07-09:
Is the font listed in `Help`->`Font List`?

If not, maybe have a look at #2511 which seems to happen on some system.

catmandu13, 2020-07-09:
You got it. It WAS missing from the Fonts list. I saw "Symbol" and took it for "Symbola". Took a bit to skim through the #2511 but the issue was....I needed to install the font as an administrator. It was accessible to other programs, but as covered by some of the other comments in #2511. I double checked my windows font list and I had some blanks in there. 

So with that and a restart of OpenSCAD:
[image]
Boom. Working! Thanks everyone!

closed 2020-07-09 by catmandu13