openscad/openscad #3371: Crashes on render of CSG with dxf rotate_extrude

RobertLipshitz, 2020-07-09:
Combining "rotate_extrude" with an "intersection" previews fine, but crashes on render (F6). Example is attached.

[crashOpenSCAD.zip](https://github.com/openscad/openscad/files/4899830/crashOpenSCAD.zip)

MichaelAtOz, 2020-07-10:
Crashes with 2019.05 & 2020.06.10 on W7/64.
On Debian 2020.06.04 it gets 
`ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation! Expr: pe != 0 File: /usr/include/CGAL/Nef_3/polyhedron_3_to_nef_3.h Line: 212 `
Changing to linear_extrude & a cube (to force CGAL) without intersection gets a different one
`ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation! Expr: e->incident_sface() != SFace_const_handle() File: /usr/include/CGAL/Nef_S2/SM_const_decorator.h Line: 326 `

OpenSCAD should not crash, it could be CGAL #2847.

But it points to bad geometry, I don't know dxf, but the line joining the curve and straight bits could be the problem??

MichaelAtOz, 2020-07-10:
Also crashed with latest 2020.07.05 on W7/64.

MichaelAtOz, 2020-07-10:
Close up of the DXF 
[image]
I suspect those corner points would turn non-manifold in 3D.

MichaelAtOz, 2020-07-10:
When I delete the right side straight bits, it doesn't get CGAL assertions.

RobertLipshitz, 2020-07-10:
I agree -- thanks. With an improved dxf (attached) it no longer crashes on render.

[doesntCrash.dxf.zip](https://github.com/openscad/openscad/files/4900537/doesntCrash.dxf.zip)

MichaelAtOz, 2020-07-10:
2015.03-2 on W7/64 (with CGAL 4.5) gets the CGAL error.
My Debian has 4.9, so I'm saying this crash is due to CGAL 4.11 as per #2847.
Closing this issue.

closed 2020-07-10 by MichaelAtOz