openscad/openscad #3373: Feature request: autosave

AiyionPrime, 2020-07-12:
Over the years openscad became a pretty stable tool, which does not crash often enough to be irritating.
As it sometimes does, though: [Bugs labeled with 'crash'](https://github.com/openscad/openscad/issues?q=is%3Aissue+is%3Aopen+label%3A%22Bug%3A+Crash%22), not even mentioning externally induced faults like an empty battery or something, it might be a good idea to implement an anonymous autosave.

scad-files are hilariously small, having an autosave every minute or so, would not hurt.
A simple filename containing a datetimestring as filename should be sufficient;
Keeping about ten of the files as well.

What do you think?

t-paul, 2020-07-12:
Note there's already a backup file created when rendering an unsaved file. See [FAQ entry](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#OpenSCAD_crashed/was_killed,_are_my_unsaved_changes_lost?).

AiyionPrime, 2020-07-12:
Mhm. In that case an integration into the UI like 'restore after crash' may be worthful to users?

akovez, 2025-04-17:
I've just lost 2h of work because openscad crashed randomly. Please, make autosave enabled by default, or autosave into backup file like {filename}.scad.backup

t-paul, 2025-04-17:
@akovez You did read the linked FAQ?