openscad/openscad #3374: Render not displaying (but is working)

gazhay, 2020-07-13:
Using latest dev build (although prev dev builds also exhibit this issue)
On Ubuntu 20.04 with AMD proprietary driver.

Simple difference object - preview
[image]
Render initially shows
[image]
Rotate render
[image]

I *think* this is due to the amd proprietary driver, however, I need the opencl capabilities of that so can't go to the foss equivalent.

If I export the render as STL it is complete in Cura or other slicers.

t-paul, 2020-07-13:
Please add the output of `Help`->`Library Info`.

gazhay, 2020-07-14:
````
OpenSCAD Version: 2020.07.05.ai5495 (git ad5f91e)
System information: Linux 5.4.0-40-generic #44-Ubuntu SMP Tue Jun 23 00:01:04 UTC 2020 x86_64 Ubuntu 20.04 LTS 32 CPUs 78.57 GB RAM
User Agent: OpenSCAD/2020.07.05.ai5495 (git ad5f91e) (Linux x86_64; Ubuntu 20.04 LTS)
Compiler: GCC "5.4.0 20160609" 64bit
MinGW build: No
Debug build: No
Boost version: 1_58
Eigen version: 3.2.92
CGAL version, kernels: 4.11, Cartesian, Extended_cartesian, Epeck
OpenCSG version: unknown, 
Qt version: 5.12.3
QScintilla version: 2.9.1
InputDrivers: 
GLib version: 2.48.2
lodepng version: 20180910
libzip version: 1.0.1
fontconfig version: 2.11.94 (runtime: 2.13.1)
freetype version: 2.6.1 (runtime: 2.10.1)
harfbuzz version: 1.0.1 (runtime: 2.6.4)
lib3mf version: 1.8.1
Features: input-driver-dbus, function-literals, lazy-union, mouse-selection
Application Path: /tmp/.mount_OpenSClpROuG/usr/bin
Documents Path: /home/gaz/.local/share
User Documents Path: /home/gaz
Resource Path: /tmp/.mount_OpenSClpROuG/usr/share/openscad
User Library Path: /home/gaz/.local/share/OpenSCAD/libraries
User Config Path: /home/gaz/.config/OpenSCAD
Backup Path: /home/gaz/.local/share/OpenSCAD/backups
OPENSCADPATH: 
OpenSCAD library path:
/home/gaz/.local/share/OpenSCAD/libraries
/tmp/.mount_OpenSClpROuG/usr/share/openscad/libraries

OPENSCAD_FONT_PATH: 
OpenSCAD font path:
/usr/share/fonts
/usr/local/share/fonts
/home/gaz/.local/share/fonts
/home/gaz/.fonts
/usr/X11/lib/X11/fonts
/System/Library/Fonts
/Library/Fonts
/home/gaz/Library/Fonts
/usr/share/fonts/X11
/usr/share/fonts/cMap
/usr/share/fonts/cmap
/usr/share/fonts/opentype
/usr/share/fonts/truetype
/usr/share/fonts/type1
/home/gaz/.fonts/Library
/home/gaz/.fonts/Local
/usr/share/fonts/X11/Type1
/usr/share/fonts/X11/encodings
/usr/share/fonts/X11/misc
/usr/share/fonts/X11/util
/usr/share/fonts/cmap/adobe-cns1
/usr/share/fonts/cmap/adobe-gb1
/usr/share/fonts/cmap/adobe-japan1
/usr/share/fonts/cmap/adobe-japan2
/usr/share/fonts/cmap/adobe-korea1
/usr/share/fonts/opentype/malayalam
/usr/share/fonts/opentype/noto
/usr/share/fonts/opentype/urw-base35
/usr/share/fonts/truetype/Gargi
/usr/share/fonts/truetype/Gubbi
/usr/share/fonts/truetype/Nakula
/usr/share/fonts/truetype/Navilu
/usr/share/fonts/truetype/Sahadeva
/usr/share/fonts/truetype/Sarai
[... 343 lines cut ...]
GL_KHR_parallel_shader_compile
GL_KHR_robust_buffer_access_behavior
GL_KHR_robustness
GL_KTX_buffer_region
GL_NV_alpha_to_coverage_dither_control
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_copy_depth_to_color
GL_NV_copy_image
GL_NV_depth_buffer_float
GL_NV_explicit_multisample
GL_NV_float_buffer
GL_NV_half_float
GL_NV_primitive_restart
GL_NV_shader_atomic_int64
GL_NV_texgen_reflection
GL_NV_texture_barrier
GL_NV_vdpau_interop
GL_OES_EGL_image
GL_SGIS_generate_mipmap
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays
GL_WIN_swap_hint
````

t-paul, 2020-07-14:
Could you please try disabling the `mouse-selection` experimental feature? This is based on some changes in the display code so maybe it's related (cc @Tomatower) - although that changes the preview, not the render display.

gazhay, 2020-07-17:
It was deselected in the UI - selecting it caused the UI to crash - when I re-opened it was enabled but caused visual glitches when selecting objects.

Tomatower, 2020-07-17:
I will try to reproduce it - though I dont have a AMD graphics card at hand.

Can you disable it in the config file, restart openscad and check if it is the cause?

`~/.config/OpenSCAD/OpenSCAD.conf`:
```
[feature]
mouse-selection=false
```

gazhay, 2020-07-30:
I have finally got round to testing with the opensource AMD driver and OpenScad renders fine.

I would suggest this isn't a bug with OpenScad but a failure of the AMD proprietary driver when installed.

Tomatower, 2020-07-30:
Have you tried it with disabled mouse-selection, in order to test if the AMD bug is somehow triggered by that feature?

gazhay, 2020-07-30:
The rendering was not altered by toggling the setting.

gazhay, 2020-08-07:
Latest version of AMDpro driver does not exhibit the issue.

Can only conclude that amd driver was responsible for the rendering issue.

closed 2020-08-07 by gazhay