openscad/openscad #3377: Polyhedrons not rendered if not hulled in some cases.

unDeadHerbs, 2020-07-19:
Polyhedrons don't exist in render when not hulled if they are passed thorough a slightly complex module.
I've produced a fairly minimal example [here](https://github.com/openscad/openscad/files/4943549/bug.txt).

Reproduced on two recent manual builds of master (562f887228bd12b749fb2d83cb2ed0ba0789d246 and 312ff6412ecb8eeefdec3ffb933644c3a054da59) and on Ubuntu (version [2019.01-RC2](https://github.com/openscad/openscad/tree/openscad-2019.01-RC2)).

t-paul, 2020-07-19:
One of the side triangles has wrong winding order, see [FAQ entry](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#Why_is_my_model_appearing_with_F5_but_not_F6?).

closed 2020-07-19 by t-paul

unDeadHerbs, 2020-07-19:
Thanks. Is it a bug that it does sometimes render then, or is this undefined behavior?

t-paul, 2020-07-19:
It's undefined what happens as the CSG libarary used (libCGAL) expects solid objects. Wong winding order means the object is not closed (manifold) so strange things can happen. `hull()` can often "fix" the issue as it's recreating a new object from a point cloud.
At some point there's hopefully better support for highlighting those kind of problems.