openscad/openscad #3378: Application crashes on start

tmilloff, 2020-07-21:
Without any changes to my system, my installation has seemingly broken without a cause. When I open up the environment, it instantly becomes unresponsive. I'm unable to interact with anything in the UI, and this is what I see:
[image]

This is some information about my system:
[image]

What I've tried:
- Reinstalling OpenSCAD
- Installing the 32 bit version instead of the 64 bit
- Installing the most recent development build
- Reinstalling the most recent drivers for my GPU

I'd be happy to provide any additional information as needed.

MichaelAtOz, 2020-07-21:
Nothing immediately stands out.
Is anything in Event Viewer?
Try setting the launcher in Regedit:
[image]
which doesn't try to do anything graphical AFAIK.

Also try taking a backup of that full OpenSCAD registry branch, then delete it.
OpenSCAD will recreate it.

tmilloff, 2020-07-21:
launcher/showOnStartup was already present and set to true. Deleting the full OpenSCAD branch fixed the issue, but I tested re-adding things from the backup, and it appears an entry in window/ was the cause:
```Windows Registry Editor Version 5.00

[HKEY_CURRENT_USER\SOFTWARE\OpenSCAD\OpenSCAD\window]
"size"="@Size(1920 1017)"
"position"="@Point(1912 -8)"
"state"=hex:40,00,42,00,79,00,74,00,65,00,41,00,72,00,72,00,61,00,79,00,28,00,\
  00,00,00,00,00,00,ff,00,00,00,00,00,00,00,00,00,fd,00,00,00,00,00,00,00,03,\
  00,00,00,00,00,00,00,00,00,00,00,00,00,01,00,f6,00,00,00,00,00,03,00,d0,00,\
  fc,00,02,00,00,00,00,00,00,00,01,00,fb,00,00,00,00,00,00,00,14,00,00,00,65,\
  00,00,00,64,00,00,00,69,00,00,00,74,00,00,00,6f,00,00,00,72,00,00,00,44,00,\
  00,00,6f,00,00,00,63,00,00,00,6b,00,01,00,00,00,00,00,00,00,15,00,00,00,00,\
  00,03,00,d0,00,00,00,00,00,00,00,7f,00,00,00,ff,00,ff,00,ff,00,00,00,00,00,\
  00,00,01,00,00,00,00,00,01,00,2f,00,00,00,00,00,03,00,d0,00,fc,00,02,00,00,\
  00,00,00,00,00,01,00,fb,00,00,00,00,00,00,00,1a,00,00,00,70,00,00,00,61,00,\
  00,00,72,00,00,00,61,00,00,00,6d,00,00,00,65,00,00,00,74,00,00,00,65,00,00,\
  00,72,00,00,00,44,00,00,00,6f,00,00,00,63,00,00,00,6b,00,01,00,00,00,00,00,\
  00,00,15,00,00,00,00,00,03,00,d0,00,00,00,00,00,00,00,9a,00,00,00,ff,00,ff,\
  00,ff,00,00,00,00,00,00,00,03,00,00,00,00,00,04,00,53,00,00,00,00,00,00,00,\
  d6,00,fc,00,01,00,00,00,00,00,00,00,01,00,fb,00,00,00,00,00,00,00,16,00,00,\
  00,63,00,00,00,6f,00,00,00,6e,00,00,00,73,00,00,00,6f,00,00,00,6c,00,00,00,\
  65,00,00,00,44,00,00,00,6f,00,00,00,63,00,00,00,6b,00,01,00,00,00,00,00,01,\
  00,fa,00,00,00,00,00,04,00,53,00,00,00,00,00,00,00,3c,00,00,00,ff,00,ff,00,\
  ff,00,00,00,00,00,04,00,53,00,00,00,00,00,02,00,f6,00,00,00,00,00,00,00,01,\
  00,00,00,00,00,00,00,02,00,00,00,00,00,00,00,01,00,00,00,00,00,00,00,02,00,\
  fc,00,00,00,00,00,00,00,00,00,29,00
```
Does this warrant further investigation, or should I close out since my original issue is resolved?

MichaelAtOz, 2020-07-21:
> Does this warrant further investigation, or should I close out since my original issue is resolved?

Leave it open. 
Given that OpenSCAD saves that info, it shouldn't save something bad, or at least detect it and not hang.

MichaelAtOz, 2020-07-21:
@tmilloff what is your screen resolution?

tmilloff, 2020-07-21:
1920 x 1080

t-paul, 2020-07-21:
That it a binary info stored by Qt, there's probably not much OpenSCAD could to here. It would be interesting to know why it's crashing though. That would need more details from the EventLog or maybe the Graphics Driver in case there's a log for that.

tmilloff, 2020-07-21:
Where in the Event Log would I find the relevant section for this? I'm not familiar with it.

t-paul, 2020-07-21:
I found this description: https://www.digitalcitizen.life/how-start-event-viewer-windows-all-versions but I'm not sure in which section the message would end up.

rcolyer, 2020-08-02:
I invested some time looking into how Qt's restoreState parses that state vector, but I couldn't find anything that should cause the observed crash.  It appears to validate byte markers as it goes through the data, and fail out with a return of false as soon as something doesn't match up to an allowed value.  The existing data looks pretty similar to other valid state vectors, and did not cause a similar crash when I converted it over and tried loading it on my system.  (Hint for posterity, the Windows approach to storing that state vector is wchar_t encoded ascii data with parentheses wrapped around raw bytes.)

@tmilloff Can you attempt to reproduce the crash by first, backing up your working config file, and then editing your config file to insert back just the "position"="@Point(1912 -8)" line?  I'd like to know if the negative value relates to the problem you had, due to something specific about your system.

tmilloff, 2020-08-03:
That position line seems to be the cause. Before removing everything to test this, the line was  "position"="@point(6 -8)" I removed everything else related to OpenSCAD from the registry, and confirmed it still ran.

Changing it to  "position"="@point(1912 -8)" caused the crash to reappear. Changing it back from 1912 to 6 fixed it again.

tmilloff, 2020-08-03:
Something appears to be going on with that x value being too high. For a given y value, there seems to be a max x value that the program can tolerate on my machine. Going above that value causes the crash on load. I tested this with a y value of around 250, and at around 1450, the program would crash every time on load. After narrowing down that value, I changed the y value back down to -8, and it appears to be somewhere around x = 1300 that the issue starts to occur.

I can get concrete values if that's helpful.

MichaelAtOz, 2020-08-03:
@tmilloff do you ever use a 2nd monitor, perhaps you had the OpenSCAD window positioned outside the main display?

tmilloff, 2020-08-05:
I do use a second monitor. Is that unsupported?

MichaelAtOz, 2020-08-05:
> unsupported

No, but obviously something is broken, it should work.

Was the failure when the 2nd monitor was not connected?

tmilloff, 2020-08-06:
No, I always run the computer with both monitors connected. I suppose it's just chance that I'd never closed the application in the 2nd window until recently.

tkotikum, 2020-11-09:
Hello everyone,

I recently started using OpenSCAD for a university project and have had this same issue on my multi-monitor setup after my first session with the software. Afterwards the default position of the environment window seems to have been on my 2nd monitor and it has resulted in it becoming unresponsive after opening any/new projects from the launcher window as described above. Multiple re-installs of different versions didn't seem to affect the issue.

However, after disconnecting all but my main monitor just now the environment is working fine now as the window is opened on my main monitor. Seems to be a multi-monitor issue then?

Let me know if I can provide any logs or additional info that would be helpful. I am quite new here :)

jordanbrown0, 2020-11-09:
I encounter this problem fairly frequently.  (Not exactly the same, because I use undocked windows, but the main view window ends up blank like this shows.)

Sometimes (but not always) when it fails I get an OpenGL popup like so:

> Warning: Missing OpenGL capabilities for OpenCSG - OpenCSG has been disabled.
>
> It is highly recommended to use OpenSCAD on a system with OpenGL 2.0 or later.
> Your renderer information is as follows:
> GLEW version 1.12.0
>  ()
> OpenGL version 

I too have a dual-monitor system.

I have a fairly repeatable case at the moment.  I start OpenSCAD, and the launcher window appears on the left (primary) monitor.  I drag it to the right monitor and click New, and I get a dead window on the right-side monitor.  Further new windows are also dead.  But if I drag one of those dead windows back over to the left side and ask for a new window, I get a good one (but still on the right side).

rcolyer, 2022-04-17:
I'll reply to @MichaelPFrey's comment from #3854 here for consolidation.

That QDesktopWidget being grabbed with QApplication::desktop has been deprecated for a long time now, and the old 4.x documentation that still describes it:  https://het.as.utexas.edu/HET/Software/html/qapplication.html#desktop  says that "_The desktop may be composed of multiple screens, so it would be incorrect, for example, to attempt to center some widget in the desktop's geometry._"  That thing marked incorrect appears to be exactly what is attempted.  So that whole Q_OS_WIN block around line 657 of src/gui/MainWindow.cc should be rewritten to something smarter, or perhaps just purged.  Does Windows even still exhibit the multi-monitor lost-window bug that was attempting to fix?

bgs-v, 2024-09-15:
Thought I was going a bit crazy, glad I found this thread and its workaround. Basically the same thing happened to me tonight. Windows often moves my windows around when my monitors come on because the left and right monitors come on nearly instantly, but the middle "main" monitor has a ~2 second long splash screen that can't be disabled (thanks ASUS) which often leads Windows to shift entire collections of windows to other monitors. My OpenSCAD just stopped working no matter what I tried, from restarts to reinstalls, and it came down to turning down the two monitors on the side to force it back to the middle - tadaa, worked again.