openscad/openscad #3380: Recursive functions hang interface

unDeadHerbs, 2020-07-24:
If a recursive function builds a growing argument it takes a very long time (longer than I've waited) to reach the max recursion depth and the interface freezes during this time.

```scad
function foo(cat=[])=
     foo(cat=concat(cat,[1]));

echo(foo());
```

t-paul, 2020-07-24:
Duplicate of #1195.

closed 2020-07-24 by t-paul

thehans, 2020-07-24:
Alternatively, an algorithmic improvement for` concat` from linear to constant time operation results in effective speedup of 10,000X (not an exaggeration*) in this particular case, if you want to test out one of the build artifacts from PR #3102.

*I saw this before I went to bed and let it run while I slept just for fun: `Total rendering time: 3:40:07.64` 
compared to `Total rendering time: 0:00:01.280` on my build from that PR.