openscad/openscad #3382: unexpected difference() result

andysc, 2020-07-26:
I'm trying to poke a hole through the base of an imported STL file, by subtracting a cylinder:
```
difference() {
    translate([-11.75,-7.7,0]) // move it to the origin
        import("MAKIES_Tea_Cup.stl");
    cylinder(d=5, h=5, $fn=50);
}
```
The STL is the tea cup from [https://www.thingiverse.com/thing:178668/files](https://www.thingiverse.com/thing:178668/files).

It looks fine with `F5`, but with `F6` it adds rather than subtracts, and you end up with a cup with a (pale green) cylinder at the bottom, which appears (and prints) as if I'd done `union()` instead of `difference()`

Should I be able to do this, or am I expecting too much of OpenSCAD here?

(I've also tried fixing up the mesh with `meshmixer`, although there were no errors from either OpenSCAD or my 3D printer software about it)

OpenSCAD 2019.05 on Mac Catalina 10.15.6

rcolyer, 2020-07-30:
It's most likely a defect issue with that imported stl file which the CGAL library is not able to handle.  Often it is very challenging to figure out what's wrong with an stl from an external source.  In this case, when I open it in meshlab and zoom in so that the viewing perspective is inside the tea cup handle, I see that there are more faces inside the volume.  In other words, this stl does not properly describe the surface of a volume, but has part of the outside inside of itself.  You might have trouble even finding a program that can repair that file.  While the stl format supports anything composed of triangles, valid volumes for CSG operations are a smaller set.

As for the program, when I use the stl file you are downloading, I also get the same unexpected difference result.  On the other hand, if you make this scad file:
```
translate([12, 8, 1]) cube(3);
```
then save that to an stl and swap that in for your filename in your example, you'll see the difference suddenly works fine.

andysc, 2020-07-30:
Thanks - yes, your diagnosis is good :)
I just tried that experiment with an STL generated from OpenSCAD (a more complicated one than you suggested, in fact) and it worked fine. So yes, it does look like a "not well formed CSG" problem.

On the other hand, I loaded the tea cup into TinkerCAD and poked a cylinder through the bottom of it without it breaking a sweat, so it seems not all STL importers are created equal...

But then I wouldn't swap OpenSCAD for TinkerCAD for anything else I do, so I guess it's just a matter of picking the right tool for certain jobs :)

Thanks again

closed 2020-08-01 by rcolyer