openscad/openscad #3384: Encoding Bug - Welcomescreen Open Files

DarkFragle, 2020-07-27:
Hello,
I think there is an encoding bug when you want to open a file.
If you use some special charakters (like Ä Ö Ü, etc) in the Filename, you can't open the file from the "Last used files" in the welcome screen, because the file was not found

[image]

[image]

When I open the File from the Windows explorer, there is no problem and when you open the file with the open file dialog, there is also no problem.

[image]

I think there is an encoding bug, because of the typical charakcter ;-)

best whises 
Fragle

rcolyer, 2020-07-28:
Looks like this is a Windows specific bug, probably related to the non-use of utf8 for filenames on that platform.  I tried the same filename on Linux and it works fine there.

thehans, 2020-07-28:
A while ago I was working on some fixes for various unicode issues, but never quite finished that project.  

I remember there was a setting in windows you can try, which seemed to fix a lot of them automatically, see my comment here: https://github.com/openscad/openscad/issues/2794#issuecomment-464422430

DarkFragle, 2020-07-28:
> A while ago I was working on some fixes for various unicode issues, but never quite finished that project.
> 
> I remember there was a setting in windows you can try, which seemed to fix a lot of them automatically, see my comment here: [#2794 (comment)](https://github.com/openscad/openscad/issues/2794#issuecomment-464422430)

@thehans Thank you for your comment. This setting (Your comment on [#2794 (comment)](https://github.com/openscad/openscad/issues/2794#issuecomment-464422430) ) works for me. This will be a good workaround. I think that this bug will not be so difficult to fix, but it is not an urgent one.

closed 2025-11-20 by kintel