openscad/openscad #3385: Out of bounds indexing should give warning

rcolyer, 2020-07-27:
Usage of a variable which is not defined evaluates to undef and gives a warning.  However, accessing an out of bounds value in a vector yields an undef without a warning.  This leads to a high potential for silent data corruption, which I have had lead to model corruption from deep within complicated routines.  This should be changed to provide an equivalent sort of warning with source file and line number.

ChrisCoxArt, 2022-01-18:
We have the information in value.cc /   Value operator()(const VectorType &vec, const double &idx) const , passed back up to the expression code.
And know that the undefined value is because of a bad vector index.

Should this always be an error?  Or a warning?  A preference for how to treat it could work, but would be confusing, and seems like overkill.  I suspect this would be a surprise to many people with bad files if we made it an error by default.  On the other hand, their files do have errors, and should be fixed, so they need at least a warning.
Compromise make it a warning now, then a year later make it an error.

nophead, 2022-01-18:
I rely on it being undef, which is false, rather then checking the length, which is extra code. In fact all additional warnings make my code longer and slower.

ChrisCoxArt, 2022-01-18:
First pass expr.cc
```
Value ArrayLookup::evaluate(const std::shared_ptr<const Context>& context) const {
    auto result = this->array->evaluate(context)[this->index->evaluate(context)];
    if (result.isUndefined()) {
        auto why = result.toUndefString();
        LOG(message_group::Warning,loc,context->documentRoot(),"vector %1$s: %2$s ", *array, why);
    }
	return result;
}
```

ChrisCoxArt, 2022-01-18:
And do we need similar warnings for named member lookup ( .r .g .b .a .x .y .z .w )?
This is handled in 
Value MemberLookup::evaluate(const std::shared_ptr<const Context>& context) const ,  and does no validation on the length of the array.

jordanbrown0, 2022-01-18:
For a language with a well-defined "undefined" value, returning "undefined" for an out-of-range access isn't silly.  That is, for instance, what JavaScript does.  (But not Python.)  Since it's what OpenSCAD has done for a long time, it's probably best to expect that programs will depend on it.

There has been some work done to put tracking information into undef values, so that if you later try to use them for something (e.g. arithmetic) you can get an error message that says something like "undefined value originated at an out-of-bounds array reference at foo.scad line 27".  I don't know what the status is of that work.

---
There is always a tension when you add new warnings.  Existing working programs may trigger those warnings.  Opinions may vary on whether the particular pattern is worrisome or not.  Perhaps the best answer is to have a fairly fine-grained way to select and deselect warnings.  Some of us will turn on every warning available and clean them up; others will not.

jordanbrown0, 2022-01-18:
Note that an object access to an undefined member also yields `undef`, and I don't think there's any other way to test for the presence of a particular member.

That is, `textmetrics("hello").nonexist` or `textmetrics("hello")["nonexist"]` will yield `undef`.

MichaelAtOz, 2022-01-18:
I vote for a. no warning or b. Preference warn-on-undef. KISS.

ChrisCoxArt, 2022-01-18:
95+% of the time bad vector indices are going to be an error, and we need something to help (especially beginner) users figure out the problems with their vector usage.
So my guess is that the best solution will be a preference that defaults to on, and can be disabled if desired.

jordanbrown0, 2025-03-09:
Answering my own comment from above... there *is* a way to detect a missing object member without relying on it being `undef`, and that in fact allows distinguishing "missing" from "set to `undef`".  You can use traversal in a list comprehension.

```
function ispresent(o, n) =
    len([ for (k = o) if (k == n) k ]) > 0;

fm = fontmetrics();
echo(ispresent(fm, "max"));
echo(ispresent(fm, "xxx"));
```