openscad/openscad #3386: Preview on Load, Preview on Save

robertlugg, 2020-07-27:
Request is to perform a preview when scad file is opened.  It is non-obvious in GUI tools that actually opening a file requires opening and then a "view" operation.  Instead on load at the very least trigger preview.  Ideally, preview to fit model in view.

Same with saving.  If you hit save and the preview is not up-to-date, you are misled into thinking what you see is what you got.  Instead perform a preview (ideally concurrently) with save (specifically <ctrl> S operation)

nophead, 2020-07-27:
Set the Automatic Reload and Preview option on the design menu.

robertlugg, 2020-07-28:
Thank you @nophead  and InPhase.  Yes, that option is what I'm looking for.  I will close this.  a "fit model in view" would also be nice but closing request instead.

closed 2020-07-28 by robertlugg