openscad/openscad #3391: openSCAD STL file cannot be sliced correctly

mobil750, 2020-08-03:
I try to print an object from Thingiverse designed with openSCAD. The STL object looks fine but after slicing it looks different, in my case a bigger hole in the middle is not cut out. You can see the STL and sliced object in the attached screenshots. 
For slicing I use the newest PrusaSlicer V2.2.0+.
I tried to fix the issue with different rearrangements in the code but with no improvement.
Attached you also find the openSCAD source file

Can anyone help here?

[tray-test.scad.zip](https://github.com/openscad/openscad/files/5015226/tray-test.scad.zip)
[tray-test.stl.zip](https://github.com/openscad/openscad/files/5015227/tray-test.stl.zip)

[image]
[image]

MichaelAtOz, 2020-08-03:
That STL has self-intersections (pink) & degenerate faces (brown).
[image]

They are generated by OpenSCAD, another case of what I call the 'small number problem', which is confounded by many co-linear faces/edges, further compounded by $fn=360 producing lots of skinny triangles.
You may want to let the author know.

Here is a clean STL of lower triangle count, if the slightly rougher curves are suitable.
[tray-test-II.zip](https://github.com/openscad/openscad/files/5019211/tray-test-II.zip)

[image]

mobil750, 2020-08-04:
Thanks for the hints. 

Yesterday I was able to get a correct STL after changing some code (removing for cycles and removing a a cone).

For new projects I will try to reduce the granularity.

rcolyer, 2020-08-05:
This was the result of previewing that scad file.  One should not attempt to render a design that previews like this, as it indicates a faulty design without well defined surfaces.  As a wish-list item it might be nice if OpenSCAD could somehow detect and warn about this sort of thing during a render attempt, but there is no known way to automatically repair a source file in this state.

[image]

MichaelAtOz, 2020-08-06:
Ryan,

 

I'm not sure how you got a preview like that.

 

The design does have problems with coincident faces/edges etc. and could be better, but that
doesn't usually result in anything like your blue image.

 

This is what I get (preview not render):

 

 

 

These sort of artefacts generally point to problems:

 

But nothing like what you got.

 

I suspect something is wrong with your system.

 

  _____  

 

 
<https://user-images.githubusercontent.com/1266642/89465212-b94d8c00-d73f-11ea-950e-17556f7cf635.pn
g> tray-test_preview

-
<https://github.com/openscad/openscad/issues/3391#issuecomment-669520156>  it on GitHub, or
unsubscribe
<https://github.com/notifications/beacon/AAPBVYOYKRS7MUZKB53XCG3R7HE55A5CNFSM4PTFQBXKYY3PNVWWK3TUL5
2HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOE7UBCHA.gif> 

 

This email has been checked for viruses by AVG.
https://www.avg.com

MichaelAtOz, 2020-08-06:
Interesting, I get the blue like messy result on my Linux box, the previous good one was on
Windows7...

 

  _____  

 

<https://user-images.githubusercontent.com/1266642/89465212-b94d8c00-d73f-11ea-950e-17556f7cf635.pn

<https://github.com/notifications/beacon/AAPBVYOYKRS7MUZKB53XCG3R7HE55A5CNFSM4PTFQBXKYY3PNVWWK3TUL5

https://www.avg.com

-
<https://github.com/notifications/beacon/AAPBVYO7DRAAZMUHFAGIUYTR7IZLDA5CNFSM4PTFQBXKYY3PNVWWK3TUL5
2HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOE7VK45Q.gif>

MichaelAtOz, 2020-08-06:
The Linux box has Intel graphics, known to have issues.

I set preferences/advanced to enable Force Goldfeather, it improved it but not perfect:

 

 

You may want to try that.

 

  _____  

 

<https://user-images.githubusercontent.com/1266642/89465212-b94d8c00-d73f-11ea-950e-17556f7cf635.pn

<https://github.com/notifications/beacon/AAPBVYOYKRS7MUZKB53XCG3R7HE55A5CNFSM4PTFQBXKYY3PNVWWK3TUL5

https://www.avg.com

-
<https://github.com/notifications/beacon/AAPBVYO7DRAAZMUHFAGIUYTR7IZLDA5CNFSM4PTFQBXKYY3PNVWWK3TUL5

<https://github.com/notifications/beacon/AAPBVYONFUHVTUBOG7R44GTR7I3FLA5CNFSM4PTFQBXKYY3PNVWWK3TUL5
2HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOE7VMMCY.gif>

MichaelAtOz, 2020-08-06:
Well, looks like the images in my emails went walkabout...

Here is:

Windows 7 clean preview

[image]

> These sort of artefacts generally point to problems:

[image]

> I set preferences/advanced to enable Force Goldfeather, it improved it but not perfect:

[image]

kintel, 2024-03-31:
@mobil750 We've improved STL export quite a bit. If you're still able to test this, please give the latest development snapshots a spin. Also check out Preferences->Features->manifold

Will close soon if we don't get any more response.

kintel, 2024-12-22:
Closing as fixed due to no further reports

closed 2024-12-22 by kintel