openscad/openscad #3394: stoi exception on commented string

bigntallmike, 2020-08-06:
I haven't quite isolated *why* yet, but throwing a bit of more useful debug code around the three stoi calls in the code lead me to discover that a fairly long array followed by a comment was causing openscad to die on an exception in `ParameterText::ParameterText`:

`stencil_drawing = [[[ lots of numbers ,126,127]]]; //[draw_polygon:100x100]`

If I move that comment down to the next line, the file opens fine.  I'm not sure why that text is ending up in the initialization to a ParameterText object, but it is.

(The full file can be found at https://www.thingiverse.com/thing:368375/files )

bigntallmike, 2020-08-06:
As an aside, this is the minor exception handler I added:

```diff
diff --git a/src/parameter/parametertext.cpp b/src/parameter/parametertext.cpp
index 596b98ea..b35b1452 100644
--- a/src/parameter/parametertext.cpp
+++ b/src/parameter/parametertext.cpp
@@ -9,7 +9,12 @@ ParameterText::ParameterText(QWidget *parent, ParameterObject *parameterobject,
 
 	double max=32767;
 	if(object->values->toVector().size() == 1){ // [max] format from makerbot customizer
-		max = std::stoi(object->values->toVector()[0]->toString(),nullptr,0);
+        try {
+		    max = std::stoi(object->values->toVector()[0]->toString(),nullptr,0);
+        } catch(std::invalid_argument &e) {
+            std::cerr << "Error in ParameterText::ParameterText: " << e.what() << " on input: " << object->values->toVector()[0]->toString() << std::endl;
+            throw;
+        }
 	}
 	this->lineEdit->setMaxLength(max);
```

bigntallmike, 2020-08-06:
A file containing only ```stencil_drawing = [[127]]; //[draw_polygon:100x100]``` will cause it to crash.
Replacing that line with ```stencil_drawing = [127]; //[draw_polygon:100x100]``` does not.

Having not read through the parsing code before, I'm not certain what's going on but that's a much shorter test case.

rcolyer, 2020-08-07:
Looks like this is a new type of crash introduced by f62d45d11fbc1a353118754e8ed56610a32b7d1f which was fixing a different crash.  I think it's trying to parse the contents inside the bracket as the maximum value for the parameter for the customizer, and failing out with an exception. //[1] does not crash, //[a] does.

closed 2020-08-07 by rcolyer

bigntallmike, 2020-08-08:
Awesome, update works for me.

PR #3395 (2020-08-07): Fix crash if non-integer in [max] customizer cmnt
Fixes #3394