openscad/openscad #3396: CGAL union error modifying STLs

rcmaniac25, 2020-08-07:
I'm using OpenSCAD for some 3D printing projects, specifically: RaspberryPi cases. I found a [good project](https://www.thingiverse.com/thing:3723561) that had an interesting methodology: they use OpenSCAD to modify existing STL models. Example: they have source STL files for the case, then have modifier STL files that they use to make holes in the case to offer customization.

To my issue: I had a reason to add some additional customization to the case generator, but used cube, cylinder, etc. to do the modifications. This seemed to work for the case I wanted, so I packaged it up and [uploaded the changes](https://www.thingiverse.com/thing:3723561).

Upon testing, I realized certain "case styles" didn't work with my modifications, giving me the following error:
`ERROR: CGAL error in CGALUtils::applyUnion: CGAL ERROR: assertion violation! Expr: itl != it->second.end() File: /mxe/usr/x86_64-w64-mingw32.static.posix/include/CGAL/Nef_3/SNC_external_structure.h Line: 1152`

Windows 10 x64
OpenSCAD version 2020.08.02.ci5674 (git bed95e77)

I tried on the current release (OpenSCAD version 2019.05) but it gave me a similar error (`ERROR: CGAL error in CGALUtils::applyBinaryOperator union: CGAL ERROR: assertion violation! Expr: itl != it->second.end() File: /mxe/usr/x86_64-w64-mingw32.static.posix/include/CGAL/Nef_3/SNC_external_structure.h Line: 1152`).

Customizer options (that differ from defaults) that cause the errors:
Style: `Hexagons - Two Materials` or `Slots - Single Material`
Add Screw Slot: true
Thicken Screw Post: true
Official PoE Support: true

I have not tried every single combination of options, but the main ones I figured people would use (given it's main goal is to make different styles).

I've attached the "[Customizer.zip](https://github.com/openscad/openscad/files/5039783/Customizer.zip)" that has all the necessary STL files and scad scripts. The scad script in question for the mentioned options above is "Case_Top_Color1.scad". If any questions or tests needed, let me know. I have compiled OpenSCAD from source before if need be.

MichaelAtOz, 2020-08-08:
Way too many coincident faces in that design.

e.g.
[image]
Anywhere you see the two colours intermingling is a coincident face.

MichaelAtOz, 2020-08-08:
...and z_top_base_sm.stl has self-intersections
[image]
those and some on one other corner.
If that STL was made with OpenSCAD, the problem may be linear_extrude() slices, and compounded by high $fn.

I haven't checked all the STLs.

rcmaniac25, 2020-08-08:
I do not have that (first) view, what option shows it? And I can understand the abnormality of the design. It's an attempt to modify the STLs that are generated by someone else.

MichaelAtOz, 2020-08-08:
> I do not have that (first) view, what option shows it?

\# makes the object translucent, * ignores that bit
```
/**********************************************************/
/* Case Generation                                        */
/**********************************************************/

*validation();

rotate(180, [0,1,0] ) {
    
    difference() {
        
        union() {
            
            case_mesh();
            
            #screw_post_thickner();
            
            *fan_hole_border_mesh();
            *fan_screws_border_mesh();
            *fan_rails_mesh();
            
            *cam_slot_border_mesh();
            *disp_slot_border_mesh();
            *pin_slot_border_mesh();
            
        }
        
        screw_slots();
        poe_fitting();
        
        front_mesh();
        left_mesh();
        right_mesh();
        
        fan_hole_mesh();
        fan_screws_mesh();
        
        cam_slot_mesh();
        disp_slot_mesh();
        pin_slot_mesh();
        
    }
    
}
```
With Official PoE Support ticked.

The case_mesh() has the self-intersections.

Interestingly that renders.

rcmaniac25, 2020-08-08:
Ah, was unaware that you could do markup. This is what I see.

[image]

The cylinder in red is added by me to make those posts larger (Thicken Screw Post) but then hollowed out (Add Screw Slot) and cut down as close to the case wall as possible (which shows as intersections, but once rendered, the wireframe shows no abnormalities).

rcmaniac25, 2020-08-08:
This is otherwise post render:
[image]

And wireframe:
[image]

MichaelAtOz, 2020-08-08:
Unfortunately, CGAL errors 99.9% are due to non-manifold STLs (or polyhedrons).
As the case_mesh STL has self intersections, any other object, which results in a union or difference, will likely get a CGAL assertion.

rcmaniac25, 2020-08-09:
*time delay*

I'll check the STLs to see if that fixes any issues.

MichaelAtOz, 2020-08-09:
> 
> 
> ...and z_top_base_sm.stl has self-intersections
> [image]
> those and some on one other corner.
> If that STL was made with OpenSCAD, the problem may be linear_extrude() slices, and compounded by high $fn.
> 
> I haven't checked all the STLs.

That is the 'Slots -Single Material' case STL, unforunately self-intersections are not the easiest to fix.

rcmaniac25, 2020-08-09:
Worst case, I've hand-edited meshes before. I just need to get 2 styles to work, but I'll be back if those are fixed of self-intersections and the error continues.

rcmaniac25, 2020-08-19:
Ok, so I went in and ran MeshMixer's auto-repair on z_top_base_sm.stl and that has resolved my issue. Could be a nice to have with the error to check that any imported STLs are manifold or similar, but I'll close this ticket as it was the geometry.

closed 2020-08-19 by rcmaniac25