openscad/openscad #3397: Rendering Polygon Mesh using CGAL... ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation! Expr: e->incident_sface() != SFace_const_handle() File: /mxe/usr/x86_64-w64-mingw32.static.posix/include/CGAL/Nef_S2/SM_const_decorator.h Line: 329

ozasaf, 2020-08-08:
Hi Guys,
In the middle of my project, this Error started.
Don't know why? no special detail, can someone help?
Thanks
####################################################################
module bomb2() {
     import("C:\\Users\\asafo\\Downloads\\Untitled\\Untitled\\mesh2.stl",convexity=3);
}

module ring3() {
    //difference() {
        color("blue") translate([0,0,-48.5]) cylinder(10-1/128,39-1/128,39-1/128,true,$fn=32);
        //color("green") translate([0,0,-48.5]) cylinder(10,35.5,35.5,true,$fn=16);
    //}
}

union() {
    translate([42,0,30.4]) ring3();
    rotate([0,90,0]) bomb2();
    //ring3();
}

nophead, 2020-08-08:
Most likely mesh2.stl has self intersections.

ozasaf, 2020-08-09:
Thanks Chris,
I agree, but this is not my design so.....
I don't have what to do/fix.
My main problem with this error, that stand alone part(render) is OK.
when I add something with union() or something else, I got this error
Thanks again

‫בתאריך שבת, 8 באוג׳ 2020 ב-19:11 מאת ‪Chris‬‏ <‪notifications@github.com
‬‏>:‬

nophead, 2020-08-09:
When you render it alone OpenSCAD just draws the mesh It doesn't matter if it is non-manifold as it you can always draw a triangle soup. As soon as you do a CSG operation on it it needs to be 2 manifold or CGAL will throw a cryptic error message.

ozasaf, 2020-08-09:
Thanks a lot Chris

‫בתאריך יום א׳, 9 באוג׳ 2020 ב-10:31 מאת ‪Chris‬‏ <‪notifications@github.com
‬‏>:‬

CD1010, 2020-09-21:
Ok, i put a similar  bug in for gear library failing. What does this mean "As soon as you do a CSG operation on it it needs to be 2 manifold"  All i do is create a 64 tooth gear using bosl gear() function and i get an error.
What would i do to fix?

nophead, 2020-09-21:
Post an issue to the BOSL library or use NopSCADlib.

[image]

kintel, 2024-03-31:
This is likely a duplicate, but we cannot verify as the STL file in question isn't available. Closing for now.

closed 2024-03-31 by kintel