openscad/openscad #3400: Allow OpenSCAD to open and write out text files

WillAdams, 2020-08-11:
Working on a fingerjoint box w/ relieved fingers: http://forum.openscad.org/Managed-to-model-a-finger-joint-box-with-relieved-fingers-now-how-to-actually-cut-it-Allow-OpenSCAD--td29878.html

I'd like to have a module write out calculated coordinate information and some text --- this would allow creating a G-Code file which could then cut the design out, no CAM tool necessary and with the possibility of doing G2/G3 arcs.

Many other scripted design tools I've worked with have allowed writing out files --- it's a powerful and expressive feature which shouldn't cause issues in other aspects of the program. 

Thanks for considering this!

t-paul, 2020-08-11:
There could be some dedicated export file format to help with that scenario. OpenSCAD is not a general purpose programming language, so it's unlikely to get some kind of general purpose `read()` or `write()` implmementation.

WillAdams, 2020-08-11:
Okay. I'm game --- how do I write out the coordinate information I'm calculating with the radisucut module so that I can then get those moves cut in a CNC?

Just being able to write out calculations would help a lot for debugging as well.

MichaelAtOz, 2020-08-11:
You `echo()` the information.
You can then use the [command line](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Using_OpenSCAD_in_a_command_line_environment) with `-o somefile.echo` to get the output to a file.

WillAdams, 2020-08-12:
Not really something user friendly which a typical user would be willing to do --- seems workable otherwise (assuming that one can control what gets output and that one can include the output of calculations and variables as their current numeric value) --- could we add a menu option to write out the echoed text to a file?

MichaelAtOz, 2020-08-12:
Select the console text, right-click-copy.

WillAdams, 2020-08-12:
Not something which a typical end user would be willing to do, and again, that would leave the requirements of:

 - selecting everything --- the G-Code files could be quite lengthy
 - switching to a text editor
 - pasting in the copied text
 - editing it to filter out the non G-Code stuff
 - saving under a suitable filename in the correct place

where instead we could have a simple File | Save Console Text (Filtered) == only the text added in the .scad file and Save Console Text (Complete) == all the text in the Console area

MichaelAtOz, 2020-08-12:
> Not something which a typical end user would be willing to do, 

CAM is not a typical use.

> and again, that would leave the requirements of:
> 
>     * selecting everything --- the G-Code files could be quite lengthy

Right-click-select-all

> 
>     * switching to a text editor
> 
>     * pasting in the copied text
> 
>     * editing it to filter out the non G-Code stuff
> 
>     * saving under a suitable filename in the correct place
> 

Or right-click-save-as. Yes you would need to filter, but if you use an appropriate Tag in the `echo()` that is trivial with tools readily available.

> where instead we could have a simple File | Save Console Text (Filtered) == only the text added in the .scad file > and Save Console Text (Complete) == all the text in the Console area

Personally I wouldn't support bringing OpenSCAD out of it's niche CAD specialisation.
However until someone commits their resources to do so, you have a few readily usable options.

WillAdams, 2020-08-12:
Writing out text files with user control is useful for more than just CAM --- CAM is a natural extension of CAD, and can potentially be added on with just a pair of menu commands as described and the return on that could be huge (esp. if there was a way to influence the extension used programmatically).

Thank you for considering this, and I hope someone can find time to add these functions.

t-paul, 2020-08-12:
With the current dev team, CAM is quite unrealistic, so if there's no specific suggestion of some geometry export format that could be used for further processing, I'm going to close this issue. If someone would like to work on CAM support, there's already other issues like #2017.

closed 2020-08-12 by t-paul