openscad/openscad #3401: Unhide editor for File->New

t-paul, 2020-08-11:
Suggestion from [forum post](http://forum.openscad.org/Opening-a-second-file-with-the-editor-hidden-tp29874p29881.html).

Creating a new, empty editor is useless with the editor hidden. So `File`->`New` could also force visibility, saving the user from doing that manually.

nophead, 2020-09-14:
I tried the latest snapshot and found it doesn't work from the welcome dialog. I normally open a new instance and select new from there. If I do that I don't get an editor and if a do new from the file menu I get an editor with two untitled tabs.

t-paul, 2020-09-16:
PR #3432 forces the editor to visible state for `new` and also when opening examples. The second case is debatable I guess, but right now I can't see a reason for opening an example without the editor visible.

PR #3428 (2020-09-05): Show Editor(if hidden), on opening new file
Fixes #3401
commit 0490686f (2020-09-17): Make editor visible for new file and examples in launching screen (fixes #3401).
commit 0deaffaa (2020-09-17): Merge pull request #3432 from openscad/unhide-editor-fix

Make editor visible for new file and examples in launching screen (fixes #3401).
closed 2020-09-17 by t-paul