openscad/openscad #3402: Negative dimensions

naggie, 2020-08-11:
When using various 2D and 3D shape functions, it would be convenient to specify negative dimensions in some cases, as an alternative to positive dimensions with a negative translation.

For instance, a cube that is required to finish at `x=0` axis instead of starting at `x=0`:

[image]

Could be specified as 

```
cube([50,-50,10]);
```
Instead of

```
translate([0,-50,0]) cube([50,50,10]);
```

This example is trivial, but I feel this may be useful in designs which rely upon many different datums and have a lot of CSG operations.

Currently nothing is displayed when negative dimensions are used. There is also no error.

Thanks for your time. OpenSCAD is the best.

t-paul, 2020-08-11:
This is not planned for the builtin modules, current path is adding warnings for such cases.

It could be handled via a user space module (maybe even at some point with the same name as the builtin one, right now it needs a different name or alternate construction instead of builtin-cube).

closed 2020-08-11 by t-paul

nophead, 2020-08-11:
I think you can override the built in cube with a user module that makes a polyhedron or linear_extrudes a square to avoid calling the original cube. I have my own version of sphere() that rotate extrudes a circle.

t-paul, 2020-08-11:
Yes, that's what I meant with "alternative construction".

Even slightly crazy stuff would be possible, like
```
module cube(s) { resize(s) rotate(45) sphere(1, $fn = 4); }
```
(Don't do that! :grin:)

rcolyer, 2020-08-11:
I concur it's a troubling path for the built-ins.  While it's understandable what it means for cube, it grows more sketchy when extended to other built-ins, and this introduces a weird inconsistency to have.  Here's a simple user-level implementation:

```
module NCube(size) { scale([for (s=size) sign(s)]) cube([for (s=size) abs(s)]); }
```