openscad/openscad #3403: openscad does not find shared object file (libQt5Core.so.5) on Fedora Linux

arnauldvm, 2020-08-12:
When trying to launch openscad on Fedora 3

```
openscad: error while loading shared libraries: libQt5Core.so.5: cannot open shared object file: No such file or directory
```

Installed with:

```
$ sudo yum install openscad
```

Some verifications I made:

```
$ yum whatprovides libQt5Core.so.5
...
qt5-qtbase-5.13.2-3.fc31.i686 : Qt5 - QtBase components
...
$ yum list installed | grep qt5-qtbase
qt5-qtbase.x86_64                           5.13.2-3.fc31                      @updates
...
$ whereis libQt5Core.so.5
libQt5Core.so: /usr/lib64/libQt5Core.so.5
$ uname -a
Linux SHADOW-5UHUC8K2 4.4.0-18362-Microsoft #476-Microsoft Fri Nov 01 16:53:00 PST 2019 x86_64 x86_64 x86_64 GNU/Linux
$ cat /etc/os-release 
NAME="Fedora Remix for WSL"
VERSION="31"
ID=fedoraremixforwsl
ID_LIKE=fedora
VERSION_ID=31
PLATFORM_ID="platform:f31"
PRETTY_NAME="Fedora Remix for WSL"
...
FEDORA_REMIX_VERSION=31.5.0
```

t-paul, 2020-08-12:
Please raise that issue with Fedora. Distribution builds are not maintained here.

closed 2020-08-12 by t-paul

t-paul, 2020-08-12:
Oh, one idea as it's not clear from the info. Try checking which binary is actually run, if that's actually the one installed via yum and not some older version installed by other means, e.g. I have:
```
$ type -a openscad
openscad is /usr/bin/openscad
openscad is /snap/bin/openscad
```
first one installed by distro, second one via Snap.

arnauldvm, 2020-08-12:
Thanks for the advice. I checked, but I do have only one version installed (it's actually a brand new box.)

Submitting to Fedora Community to get more advices to investigate this.

Will report here when I find something.